mpromonet/webrtc-streamer #285: Build errors on ubuntu 16.04 related to X11

rahul3349, 2019-11-22:
Error logs: 
[ 95%] Linking CXX executable webrtc-streamer
/usr/bin/ld: cannot find -lX11
/usr/bin/ld: cannot find -lXext
/usr/bin/ld: cannot find -lXdamage
/usr/bin/ld: cannot find -lXfixes
/usr/bin/ld: cannot find -lXcomposite

mpromonet, 2019-11-22:
Hi,
You may disable X11 dependency using : 

    cmake -DWEBRTCDESKTOPCAPTURE=OFF ....

Best Regards,
Michel.

rahul3349, 2019-11-23:
Thanks Michel.
I've installed dev packages of all libs and able to resolve build error related to these packages but getting more build errors related to webrtc.
There is no proper build instructions provided with project, especially related to webrtc. 
I've created new post(#286 )related to this issue, requesting you to provide detailed instruction to build webrtc and which version/commit of webrtc is compatible with this project.

closed 2019-11-23 by rahul3349