mpromonet/webrtc-streamer #286: Build error related to webrtc

rahul3349, 2019-11-23:
Getting build error while linking executable,

I used below cmd to build webrtc
gn gen out/Release --args='is_debug=false rtc_use_h264=true ffmpeg_branding="Chrome"   treat_warnings_as_errors=false rtc_include_tests=false rtc_enable_protobuf=false use_custom_libcxx=false use_ozone=true rtc_include_pulse_audio=false rtc_build_examples=false'
also buid jsoncpp to remove linking error related to it.
but still getting below build errors,.

Author of project should provide details instruction to build the project, webrtc, version of webrtc etc

~/sources/webrtc-streamer$ make
[  1%] Built target webrtcextra
[  2%] Built target civetweb
[ 95%] Built target liblive555helper
[ 95%] Linking CXX executable webrtc-streamer
CMakeFiles/webrtc-streamer.dir/src/PeerConnectionManager.cpp.o: In function `PeerConnectionManager::addIceCandidate(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Json::Value const&)':
PeerConnectionManager.cpp:(.text+0x3c0e): undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
PeerConnectionManager.cpp:(.text+0x3e6a): undefined reference to `rtc::GetIntFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int*)'
PeerConnectionManager.cpp:(.text+0x3f78): undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
CMakeFiles/webrtc-streamer.dir/src/PeerConnectionManager.cpp.o: In function `PeerConnectionManager::setAnswer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Json::Value const&)':
PeerConnectionManager.cpp:(.text+0x46ac): undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
PeerConnectionManager.cpp:(.text+0x4ae3): undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
CMakeFiles/webrtc-streamer.dir/src/PeerConnectionManager.cpp.o: In function `PeerConnectionManager::call(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Json::Value const&)':
PeerConnectionManager.cpp:(.text+0x9b70): undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
PeerConnectionManager.cpp:(.text+0xa207): undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
/home/rbh/sources/webrtc/src/out/Release/obj/libwebrtc.a(base_capturer_pipewire.o): In function `webrtc::BaseCapturerPipeWire::~BaseCapturerPipeWire()':
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0xae): undefined reference to `g_dbus_connection_signal_unsubscribe'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0xc4): undefined reference to `g_dbus_connection_signal_unsubscribe'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0xda): undefined reference to `g_dbus_connection_signal_unsubscribe'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0x104): undefined reference to `g_dbus_message_new_method_call'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0x12b): undefined reference to `g_dbus_connection_send_message'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0x162): undefined reference to `g_error_free'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0x16a): undefined reference to `g_object_unref'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0x176): undefined reference to `g_free'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0x182): undefined reference to `g_free'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0x18e): undefined reference to `g_free'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0x19a): undefined reference to `g_free'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0x1ab): undefined reference to `g_cancellable_cancel'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0x1b7): undefined reference to `g_object_unref'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWireD2Ev+0x1d3): undefined reference to `g_object_unref'
/home/rbh/sources/webrtc/src/out/Release/obj/libwebrtc.a(base_capturer_pipewire.o): In function `webrtc::BaseCapturerPipeWire::InitPortal()':
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire10InitPortalEv+0xa): undefined reference to `g_cancellable_new'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire10InitPortalEv+0x43): undefined reference to `g_dbus_proxy_new_for_bus'
/home/rbh/sources/webrtc/src/out/Release/obj/libwebrtc.a(base_capturer_pipewire.o): In function `webrtc::BaseCapturerPipeWire::OnProxyRequested(_GObject*, _GAsyncResult*, void*)':
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire16OnProxyRequestedEP8_GObjectP13_GAsyncResultPv+0x1e): undefined reference to `g_dbus_proxy_new_finish'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire16OnProxyRequestedEP8_GObjectP13_GAsyncResultPv+0x32): undefined reference to `g_dbus_proxy_get_connection'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire16OnProxyRequestedEP8_GObjectP13_GAsyncResultPv+0x6d): undefined reference to `g_io_error_quark'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire16OnProxyRequestedEP8_GObjectP13_GAsyncResultPv+0x7c): undefined reference to `g_error_matches'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire16OnProxyRequestedEP8_GObjectP13_GAsyncResultPv+0xb2): undefined reference to `g_error_free'
/home/rbh/sources/webrtc/src/out/Release/obj/libwebrtc.a(base_capturer_pipewire.o): In function `webrtc::BaseCapturerPipeWire::SetupRequestResponseSignal(char const*, void (*)(_GDBusConnection*, char const*, char const*, char const*, char const*, _GVariant*, void*))':
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire26SetupRequestResponseSignalEPKcPFvP16_GDBusConnectionS2_S2_S2_S2_P9_GVariantPvE+0x37): undefined reference to `g_dbus_connection_signal_subscribe'
/home/rbh/sources/webrtc/src/out/Release/obj/libwebrtc.a(base_capturer_pipewire.o): In function `webrtc::BaseCapturerPipeWire::SessionRequest()':
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire14SessionRequestEv+0x59): undefined reference to `g_variant_builder_init'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire14SessionRequestEv+0x65): undefined reference to `g_random_int_range'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire14SessionRequestEv+0x75): undefined reference to `g_strdup_printf'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire14SessionRequestEv+0x80): undefined reference to `g_variant_new_string'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire14SessionRequestEv+0x9e): undefined reference to `g_variant_builder_add'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire14SessionRequestEv+0xa6): undefined reference to `g_free'
base_capturer_pipewire.cc:(.text._ZN6webrtc20BaseCapturerPipeWire14SessionRequestEv+0xb2): undefined reference to `g_random_int_range'
[... cut ...]
stun_server.cc:(.text._ZNK7cricket10StunServer19GetStunBindResponseEPNS_11StunMessageERKN3rtc13SocketAddressES2_+0x99): undefined reference to `cricket::StunMessage::AddAttribute(std::__1::unique_ptr<cricket::StunAttribute, std::__1::default_delete<cricket::StunAttribute> >)'
collect2: error: ld returned 1 exit status
CMakeFiles/webrtc-streamer.dir/build.make:280: recipe for target 'webrtc-streamer' failed
make[2]: *** [webrtc-streamer] Error 1
CMakeFiles/Makefile2:143: recipe for target 'CMakeFiles/webrtc-streamer.dir/all' failed
make[1]: *** [CMakeFiles/webrtc-streamer.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

mpromonet, 2019-11-23:
Hi @rahul3349 

This is an unlicensed project, you are free to use it and to improve it.
Main information for building are in the readme : 

```
mkdir ../webrtc
pushd ../webrtc
fetch --no-history webrtc 
popd
cmake . 
make
```

You can get more details from automatic build, for instance from https://github.com/mpromonet/webrtc-streamer/blob/master/Dockerfile

I guess the problem you have, is because you run gn but not ninja (and thus not build webrtc)
If you follow the readme instruction and not run gn, cmake will do it. 
Removing webrtc/rtc/out and running cmake again should helps you.

Best Regards,
Michel.

rahul3349, 2019-11-24:
@mpromonet 
Yes, initially I followed instructions given in readme, but could not able build webrtc using cmake cmd, so I was trying to build it manually. But as you mentioned that webrtc should be build only using cmake so I gave try again, also tried to resolve build error in webrtc, now I'm able to build it successfully, also able to run webrtc-streamer.
Thanks for your support.  This is really a great project.

closed 2019-11-24 by rahul3349

mpromonet, 2019-11-24:
@rahul3349 
You are welcome.
This project is experiments trying to understand better webrtc technology.
Feel free to improve it.
Best Regards,
Michel.