mpromonet/webrtc-streamer #288: The video not displayed in player

MashaFomina, 2019-11-29:
I can not run demo on my local machine with docker image. My host OS - Windows 10. 
Maybe someone has the same problem?

I run docker image with command `docker run -vvv -p 8000:8000 -it mpromonet/webrtc-streamer`. The content of logs:
```
{
        "urls" :
        {
                "Arrifana" :
                {
                        "options" : "rtptransport=tcp&timeout=60",
                        "video" : "rtsp://46.105.54.176:80/64connections/arrifana.stream"
                },
        ...
}
Logger level:3
[000:000] [1] (adm_helpers.cc:44): Unable to set playout device.
[000:001] [1] (audio_device_generic.cc:18): BuiltInAECIsAvailable: Not supported on this platform
[000:002] [1] (audio_device_generic.cc:28): BuiltInAGCIsAvailable: Not supported on this platform
[000:002] [1] (audio_device_generic.cc:38): BuiltInNSIsAvailable: Not supported on this platform
HTTP Listen at 0.0.0.0:8000
uri:/api/getIceServers
answer:{
        "iceServers" :
        [
                {
                        "urls" :
                        [
                                "stun:stun.l.google.com:19302"
                        ]
                }
        ]
}
uri:/api/version
answer:"5480f27/Linux-x86_64 civetweb@v1.11 webrtc@51868f52c6-dirty live555helper@582732b"
uri:/api/call
[017:493] [1] (PeerConnectionManager.h:223): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.7019852003720202
[017:499] [1] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.7019852003720202
[017:502] [1] (PeerConnectionManager.h:223): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.7019852003720202
[017:508] [1] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.7019852003720202
answer:{
        "sdp" : "v=0\r\no=- 7244206058668757391 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE 0 1 2\r\na=msid-semantic: WMS 15961245695042183131\r\nm=audio 49489 UDP/TLS/RTP/SAVPF 111 103 104 9 0 8 106 105 13 110 112 113 126\r\nc=IN IP4 172.17.0.2\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=candidate:4112619227 1 udp 2122260223 172.17.0.2 49489 typ host generation 0 network-id 1\r\na=ice-ufrag:/opY\r\na=ice-pwd:ErUcZZsns9+J90q0CbmGCtps\r\na=ice-options:trickle\r\na=fingerprint:sha-256 06:71:30:74:7A:50:13:B9:D5:39:89:69:95:8C:A1:C5:64:10:ED:B9:2B:34:E5:2F:9E:81:68:F2:39:29:3E:2D\r\na=setup:active\r\na=mid:0\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:3 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=sendonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:104 ISAC/32000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:106 CN/32000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:112 telephone-event/32000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\na=ssrc:2988287911 cname:y0l0VR90o3njDADR\r\na=ssrc:2988287911 msid:15961245695042183131 15961245695042183131_audio\r\na=ssrc:2988287911 mslabel:15961245695042183131\r\na=ssrc:2988287911 label:15961245695042183131_audio\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 100 101 102 122 127 121 125 107 108 109 114 115 116\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:/opY\r\na=ice-pwd:ErUcZZsns9+J90q0CbmGCtps\r\na=ice-options:trickle\r\na=fingerprint:sha-256 06:71:30:74:7A:50:13:B9:D5:39:89:69:95:8C:A1:C5:64:10:ED:B9:2B:34:E5:2F:9E:81:68:F2:39:29:3E:2D\r\na=setup:active\r\na=mid:1\r\na=extmap:14 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:13 urn:3gpp:video-orientation\r\na=extmap:3 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:12 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=extmap:11 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type\r\na=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-timing\r\na=extmap:8 http://tools.ietf.org/html/draft-ietf-avtext-framemarking-07\r\na=extmap:9 http://www.webrtc.org/experiments/rtp-hdrext/color-space\r\na=sendonly\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:96 VP8/90000\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:98 VP9/90000\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=fmtp:98 profile-id=0\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:100 VP9/90000\r\na=rtcp-fb:100 goog-remb\r\na=rtcp-fb:100 transport-cc\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=fmtp:100 profile-id=2\r\na=rtpmap:101 rtx/90000\r\na=fmtp:101 apt=100\r\na=rtpmap:102 H264/90000\r\na=rtcp-fb:102 goog-remb\r\na=rtcp-fb:102 transport-cc\r\na=rtcp-fb:102 ccm fir\r\na=rtcp-fb:102 nack\r\na=rtcp-fb:102 nack pli\r\na=fmtp:102 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42001f\r\na=rtpmap:122 rtx/90000\r\na=fmtp:122 apt=102\r\na=rtpmap:127 H264/90000\r\na=rtcp-fb:127 goog-remb\r\na=rtcp-fb:127 transport-cc\r\na=rtcp-fb:127 ccm fir\r\na=rtcp-fb:127 nack\r\na=rtcp-fb:127 nack pli\r\na=fmtp:127 level-asymmetry-allowed=1;packetization-mode=0;profile-level-id=42001f\r\na=rtpmap:121 rtx/90000\r\na=fmtp:121 apt=127\r\na=rtpmap:125 H264/90000\r\na=rtcp-fb:125 goog-remb\r\na=rtcp-fb:125 transport-cc\r\na=rtcp-fb:125 ccm fir\r\na=rtcp-fb:125 nack\r\na=rtcp-fb:125 nack pli\r\na=fmtp:125 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na=rtpmap:107 rtx/90000\r\na=fmtp:107 apt=125\r\na=rtpmap:108 H264/90000\r\na=rtcp-fb:108 goog-remb\r\na=rtcp-fb:108 transport-cc\r\na=rtcp-fb:108 ccm fir\r\na=rtcp-fb:108 nack\r\na=rtcp-fb:108 nack pli\r\na=fmtp:108 level-asymmetry-allowed=1;packetization-mode=0;profile-level-id=42e01f\r\na=rtpmap:109 rtx/90000\r\na=fmtp:109 apt=108\r\na=rtpmap:114 red/90000\r\na=rtpmap:115 rtx/90000\r\na=fmtp:115 apt=114\r\na=rtpmap:116 ulpfec/90000\r\na=ssrc-group:FID 3833343206 3826122310\r\na=ssrc:3833343206 cname:y0l0VR90o3njDADR\r\na=ssrc:3833343206 msid:15961245695042183131 15961245695042183131_video\r\na=ssrc:3833343206 mslabel:15961245695042183131\r\na=ssrc:3833343206 label:15961245695042183131_video\r\na=ssrc:3826122310 cname:y0l0VR90o3njDADR\r\na=ssrc:3826122310 msid:15961245695042183131 15961245695042183131_video\r\na=ssrc:3826122310 mslabel:15961245695042183131\r\na=ssrc:3826122310 label:15961245695042183131_video\r\nm=application 9 UDP/DTLS/SCTP webrtc-datachannel\r\nc=IN IP4 0.0.0.0\r\nb=AS:30\r\na=ice-ufrag:/opY\r\na=ice-pwd:ErUcZZsns9+J90q0CbmGCtps\r\na=ice-options:trickle\r\na=fingerprint:sha-256 06:71:30:74:7A:50:13:B9:D5:39:89:69:95:8C:A1:C5:64:10:ED:B9:2B:34:E5:2F:9E:81:68:F2:39:29:3E:2D\r\na=setup:active\r\na=mid:2\r\na=sctp-port:5000\r\na=max-message-size:262144\r\n",
        "type" : "answer"
}
uri:/api/addIceCandidate
uri:/api/addIceCandidate
uri:/api/addIceCandidate
uri:/api/addIceCandidate
uri:/api/getIceCandidate
answer:true
answer:true
answer:true
answer:true
answer:[
        {
                "candidate" : "candidate:4112619227 1 udp 2122260223 172.17.0.2 49489 typ host generation 0 ufrag /opY network-id 1",
                "sdpMLineIndex" : 0,
                "sdpMid" : "0"
        }
]
Created new TCP socket 62 for connection
Created new TCP socket 61 for connection
[019:906] [64] (liveaudiosource.h:98): LiveAudioSource::onNewSession not support codecm=audio 0 RTP/AVP 96
a=rtpmap:96 mpeg4-generic/12000/2
a=fmtp:96 profile-level-id=1;mode=AAC-hbr;sizelength=13;indexlength=3;indexdeltalength=3;config=1490
a=control:trackID=1

Start playing sink for "video/H264" subsession
markerSize:4
[032:613] [69] (PeerConnectionManager.cpp:735): Remove PeerConnection peerid:0.7019852003720202
[032:613] [69] (PeerConnectionManager.cpp:752): hangUp stream is no more used 15961245695042183131
[032:614] [69] (PeerConnectionManager.cpp:760): hangUp stream closed 15961245695042183131
[... cut ...]
webrtcstreamer.js:279 candidate: [{"candidate":"candidate:4112619227 1 udp 2122260223 172.17.0.2 36435 typ host generation 0 ufrag GFPB network-id 1","sdpMLineIndex":0,"sdpMid":"0"}]
webrtcstreamer.js:284 Adding ICE candidate :{"candidate":"candidate:4112619227 1 udp 2122260223 172.17.0.2 36435 typ host generation 0 ufrag GFPB network-id 1","sdpMid":"0","sdpMLineIndex":0}
webrtcstreamer.js:227 addIceCandidate ok:true
webrtcstreamer.js:160 oniceconnectionstatechange  state: checking
webrtcstreamer.js:286 addIceCandidate OK
webrtcstreamer.js:217 End of candidates.
webrtcstreamer.js:160 oniceconnectionstatechange  state: disconnected
```

mpromonet, 2019-11-30:
Hi MashaFomina,

It may be a network configuration problem, inside the container should resolve `stun.l.google.com` and the some ICE candidate should be accessible.
It depends on docker DNS configuration docker network configuration.
You may troubleshooting analysing the log produced starting the container with : 

    docker run  -p 8000:8000 -it mpromonet/webrtc-streamer -C config.json -vvv

Best Regards,
Michel.

MashaFomina, 2019-12-01:
Hi Michel. Thank you for your response.
I tried to run image with flags `-vvv`. I also checked access to `stun.l.google.com` in the docker container.
```
$ ping stun.l.google.com
PING stun.l.google.com (64.233.165.127) 56(84) bytes of data.
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=1 ttl=40 time=22.0 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=2 ttl=40 time=24.0 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=3 ttl=40 time=26.8 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=4 ttl=40 time=25.9 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=5 ttl=40 time=21.8 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=6 ttl=40 time=19.5 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=7 ttl=40 time=19.7 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=8 ttl=40 time=20.2 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=9 ttl=40 time=22.1 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=10 ttl=40 time=25.2 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=11 ttl=40 time=19.9 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=12 ttl=40 time=22.7 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=13 ttl=40 time=20.2 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=14 ttl=40 time=20.0 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=15 ttl=40 time=20.1 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=16 ttl=40 time=19.8 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=17 ttl=40 time=23.3 ms
64 bytes from lg-in-f127.1e100.net (64.233.165.127): icmp_seq=18 ttl=40 time=21.8 ms
^C
--- stun.l.google.com ping statistics ---
18 packets transmitted, 18 received, 0% packet loss, time 17049ms
rtt min/avg/max/mdev = 19.547/21.994/26.836/2.238 ms
```
I checked the logs and I did not see any problems in it. 
[webrtc-streamer-log.txt](https://github.com/mpromonet/webrtc-streamer/files/3907748/log.txt)

I think that my problem can be related with client side (`webrtcstreamer.html`).

Best Regards,
Mariia.

projscorpio, 2019-12-02:
Got same problem.

mpromonet, 2019-12-03:
Did you tried without docker, just running the executable from https://github.com/mpromonet/webrtc-streamer/releases ?

MashaFomina, 2019-12-03:
Hi Michel.  I downloaded release for windows on my local machine and it works properly, displays  video.

Best Regards,
Mariia.

mpromonet, 2019-12-05:
Hi Mariia,

Then I guess it is a network configuration of docker.
From the logs, we see RTP packets are received, but it seems no ICE candidate is selected.
This may depends on the docker network driver used.

Best Regards,
Michel.

MashaFomina, 2019-12-05:
Hi Michel. 

Thank you for response. I resolved the problem by using host network driver.
```
# command for Windows 10, use host network driver
docker run --rm -p 8000:8000 --network host -it mpromonet/webrtc-streamer
```

Best Regards,
Mariia.

closed 2019-12-05 by MashaFomina