mpromonet/webrtc-streamer #289: What type of license of civetweb applicable for this project?

rahul3349, 2019-12-05:
What type of license of civetweb applicable for this project?
in civetweb project, they have different licenses and mentioned it is applicable on usage based.
https://github.com/civetweb/civetweb/blob/ce8f6d38a60eb16c996afee1e5340f76ef4d0923/LICENSE.md

mpromonet, 2019-12-05:
Hi,

First I am not a layer, I am sharing some experiments under [Unlicense](https://unlicense.org/) and people that would like to improve are welcomed.

My understanding is that you can do whatever you like under your responsability, there is no support, no waranty of any kind, and my responsabilty is not engaged by use that you could do.

The file https://github.com/civetweb/civetweb/blob/ce8f6d38a60eb16c996afee1e5340f76ef4d0923/LICENSE.md appear to me as the license of each pieces depending on build option, this is not different license about civetweb that is an "AS IS" license.

WebRTC is also an "AS IS" license https://webrtc.org/license/software/

live555 is under LGPL that is, I think, compliant with the "AS IS" license of this project, because the end-user have all the material to upgrade it, source code and build chain.

Best Regards,

Michel.

rahul3349, 2019-12-12:
@mpromonet Thanks for clarification

closed 2019-12-12 by rahul3349