mpromonet/webrtc-streamer #291: arm64 build failed

mmilunovic, 2019-12-16:
Hi,

I am trying to build a webrtc-streamer on Jetson Nano or Jetson TX2. 
Both are running the latest NVIDIA Jetpack, which is basically an Ubuntu 18.04 for Jetson devices. 

Then I try to run:

```
fetch --no-history webrtc
```

I get:

```
Host architecture arm64 is not supported.
```
I tried to use the Docker image provided in the instructions and that also doesn't work. 

```
docker run -p 8000:8000 -it mpromonet/webrtc-streamer
```
```
Unable to find image 'mpromonet/webrtc-streamer:latest' locally
latest: Pulling from mpromonet/webrtc-streamer
docker: no matching manifest for linux/arm64/v8 in the manifest list entries.

```

**Is there a way to build on the Jetson's?**

mpromonet, 2019-12-19:
Hi mmilunovic,
I never try to build for arm64, but you may have a similar approach for arm32 crosscompiling it on x86-64.
Best Regards,
Michel.

mmilunovic, 2019-12-19:
> Hi mmilunovic,
> I never try to build for arm64, but you may have a similar approach for arm32 crosscompiling it on x86-64.
> Best Regards,
> Michel.

Hi Michael, 

I'm not sure that I understand what you're saying?
Could you point me to some sort of a instruction?

Best, 
Milos

mpromonet, 2019-12-19:
Hi Milos,

Maybe you can start with commands in https://github.com/mpromonet/webrtc-streamer/blob/master/Dockerfile.rpi adapting with a crosscompiler for Jetson (it seems to be available from https://developer.nvidia.com/embedded/dlc/l4t-gcc-toolchain-64-bit-28-3).
It may also some modification in the CMakefile to build WebRTC SDK, the options depends on platform.
Best Regards,
Michel.

mmilunovic, 2019-12-19:
Hi Michael, 

Not sure I'm being clear, I get the same error when I run Docker on the regular build. 
The problem happens when I try to install WebRTC, not build WebRTC Streamer.

mpromonet, 2019-12-19:
Hi Milos,
I add a docker that compile arm64 using cross-compiler `aarch64-linux-gnu`, You probably could inspired from this to build with the toolchain corresponding to your board.
Best Regards,
Michel.

closed 2020-02-13 by mpromonet