mpromonet/webrtc-streamer #293: Build on windows 7 failed

laowangsir, 2019-12-26:
I got an error message from cmake   **fatal error: 'pthread.h' file not found**

environment  ： windows 7  + LLVM7.0 + VS2017 + CMAKE 3.16 + depot_tools
I cannot access webrtc code from google site ,so   I download it from github 
(https://github.com/JumpingYang001/webrtc)  and put it on webrtc-stream/../webrtc
the execute  cmake -G Ninja -DCMAKE_C_COMPILER=clang-cl.exe -DCMAKE_CXX_COMPILER=clang-cl.exe .      and it failed

laowangsir, 2019-12-26:
The second step is F:\vs2017\VC\Auxiliary\Build\vcvarsamd64_x86.bat

mpromonet, 2020-01-02:
Hi laowangsir,
What do you means by download webrtc from github ? 
See WebRTC SDK documentation https://webrtc.org/native-code/development/
You may get it using depot_tools.
Best Regards,
Michel.

laowangsir, 2020-01-04:
Thank u mpromonet  . I  finally built it using depot_tools and VPN . But I dont know how to use this app ...  Does it works as a stream server , pull stream and push stream in HTTP? 
webrtc-streamer rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov
what does the command mean ?

mpromonet, 2020-01-04:
Hi,
In case of RTSP, it is a gateway implementing an RTSP consumer and a WebRTC producer.
You can start with the demo app https://webrtc-streamer.herokuapp.com/.
Running webrtc-streamer with an rtsp url, you should be able to access to WebRTC stream connecting to http://...:8000.
Best Regards,
Michel.

laowangsir, 2020-01-04:
Thanks . In an video chat case , a IPCam push stream to video server , and webrtc-streamer pull stream from the video server and push a webrtc stream ?   I'm working on a program that broadcast screen capture of PC (win)  ,and I saw webrtc-streamer support the command  webrtc-streamer  screen://xx.host.xxx  ?  What does it mean (the parameter after screen://) ,I think the screen of the host running webrtc-streamer should be record and broadcast ,not the host following screen:// . 
BTW, I read the code , this feature is only available when #ifdef USE_X11 is defined . That means that windows  has not the feature ?

mpromonet, 2020-01-04:
Hi,

The screen syntax `screen://<id of the screeen>`. If screen & windows feature is enabled, you can get the list of possible url using /api/getMediaList. These url allow to stream through WebRTC windows were webrtc-streamer run (as it could stream captrue device like webcam)..   

The flag should be better named, on windows the feature is enabled. If you look it Cmakefile : 
```add_definitions(-DUSE_X11 -DHAVE_SOUND -D_WINSOCKAPI_)```
 
Best Regards,
Michel.

laowangsir, 2020-01-06:
THANKS mpromonet!    I've tried screen capture feature.,and got the stream!    My steps ： 1 Add an entry to config.json  "scr": {"video": "screen://myhostname"}  and launch the app with command webrtc-streamer     screen://myhostname  .   I dont known if there are other parameters available . 
In the browser , I found only one window is captured . If I could get the whole screen ? should I modify the source code or command line ?

laowangsir, 2020-01-07:
Lol.  I  took a look at the source and find that screen://-1  means full screen capture. I execute streamer with this parameter , it worked . Still there is some other video source in  menu entry.

[image]

laowangsir, 2020-01-07:
I've modified  function  CapturerFactory::GetVideoSourceList , delete the section that create window capture , now there's only one window ....  
two more  question 1 .    I want the stream to be real time , no buffer , no delay 
                                 2.     I want to shrink the image size of the video stream (need to display on android pad )
Is this possible ? thx

mpromonet, 2020-01-07:
Hi,
Maybe using `-q screen://` could help you to disable the window:// url.

Some hints about your questions : 
1. There is no buferring more than in WebRTC SDK
2. the options width & height allow you to resize an webrtc stream
3. the option bitrate allow to limit bitrate between value/2 and value*2

Best Regards,
Michel.

laowangsir, 2020-01-09:
Thank u , the parameter -q works.  Now I test webrtc-streamer in WLAN environment , got some  error message . (I add some log in webrtc ) 

1. (stun_port.cc:308): Port[af5e0:0:1:0:local:Net[Intel(R):192.168.0.x/32:Wifi:id=3]]: UDP send of 1207 bytes failed with error 10035
2.  [032:304] [16532] (stun_port.cc:596): sendto  stun.l.google.com:19302 : [0x00002743] A socket operation was attempted to an unreachable network.

closed 2020-01-12 by mpromonet