mpromonet/webrtc-streamer #294: Streaming fil syntax?

rossbcan, 2019-12-30:
Everything working except Bunny. Is my file:/// syntax correct?

demo site: www.rossco.org:8000

{
    "urls":{
        "Patio": {"video":"rtsp://192.168.10.44:554/11"},
        "Antenna": {"video":"rtsp://192.168.10.46/11"},
        "Front-Left": {"video":"rtsp://192.168.10.48:8554/0"},
        "Front-Right": {"video":"rtsp://192.168.10.49:8554/0"},
        "Bunny": {"video":"file:///home/data/videos/test1.mkv"},
        "VirginiaTech": {"video":"rtsp://198.82.159.136:554/axis-media/media.amp"}
    }
}

Thanks;
Bill

mpromonet, 2020-01-02:
Hi Bill,
The syntax `"Bunny": {"video":"file:///home/data/videos/test1.mkv"}` is ok to stream the file `/home/data/videos/test1.mkv` 
It needs the mkv to use a supported codec (H264, VP9, JPEG).
Running with `-vvv` you should get more information why it is not working.
Best Regards
Michel.

rossbcan, 2020-01-03:
Hi Michael;

Changed url of test site to www.rossco.org/cameras
Instructions regarding proxying webrtc-streamer here: https://forum.nginx.org/read.php?11,286648
Any feedback re nginx proxy config welcome 

I used vlc to save and transcode audio from: rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov to use opus audio. Changed config for bunny file only.

ipv6 disabled for now

webrtc-streamer -H 0.0.0.0:8000 -w /usr/webrtc-streamer -N 2 -sstun.l.google.com:19302 -a3 -C /var/etc/webrtc-streamer.conf
log: https://www.rossco.org/Downloads/OpenWrt/Files/webrtc-streamer.txt
salient points:
(stun_port.cc:595): sendto : [0x0000000D] Permission denied
ERROR: File does not begin with an EBML header

webrtc-streamer -vvv -H 0.0.0.0:8000 -w /usr/webrtc-streamer -N 2 -sstun.l.google.com:19302 -a3 -C /var/etc/webrtc-streamer.conf
log: https://www.rossco.org/Downloads/OpenWrt/Files/webrtc-streamer-full.txt
salient points:
as above
Unsupported RTP extension: {uri: urn:ietf:params:rtp-hdrext:csrc-audio-level, id: 2}
Not adding candidate because the JsepTransport doesn't exist. Ignore it.

Receiving no audio or video. Not sure why, or the importance of the various errors in logs (translation to realspeak: unsure what is wrong, how to fix).

It would really help if I had a bunny file with known good audio / video from webrtc-streamer perspective.

My use case is providing easy / safe IP camera / video view ability for OpenWrt community.

Another question: I have ssl key and cert. How to I use them with sslkeycert config value?

Thanks;
Bill

mpromonet, 2020-01-03:
Hi Bill,

The error `ERROR: File does not begin with an EBML header` says the format is not a valid mkv (or not parsable by live555).

You can try (for video only) with file https://test-videos.co.uk/vids/bigbuckbunny/webm/vp9/360/Big_Buck_Bunny_360_10s_1MB.webm 

Best Regards,
Michel.

closed 2020-01-08 by rossbcan