mpromonet/webrtc-streamer #297: Delay tween your checkin and git?

rossbcan, 2020-01-08:
Building with latest aspect ratio preserve fix.
Is there are time delay between your checkin and git?
All I did was change version...

Checking out files from the git repository...
Cloning into 'webrtc-streamer-48039cbd5e6faeb72ea83b721f0bbf47788fc7a3'...
remote: Enumerating objects: 51, done.
remote: Counting objects: 100% (51/51), done.
remote: Compressing objects: 100% (38/38), done.
remote: Total 5226 (delta 21), reused 36 (delta 12), pack-reused 5175
Receiving objects: 100% (5226/5226), 5.77 MiB | 7.36 MiB/s, done.
Resolving deltas: 100% (3425/3425), done.
fatal: reference is not a tree: 48039cbd5e6faeb72ea83b721f0bbf47788fc7a3

Regards;
Bill

mpromonet, 2020-01-09:
Hi Bill,

What do you call delay between checkin and git ?
Maybe if you describe the command you are using it could be clearer.
Anyway commit 48039cbd5e6faeb72ea83b721f0bbf47788fc7a3 is not on webrtc-streamer but on webrtc-streamer-html.

Best Regards,
Michel.

rossbcan, 2020-01-10:
You are correct. I incorrectly applied webrtc-streamer-html git version to webrtc-streamer. oops...

closed 2020-01-10 by rossbcan