mpromonet/webrtc-streamer #298: CPU load very high when stream video RTSP by webrtc

nokiovo79, 2020-01-12:
There is a way to avoid frame parsing and send the copy of the rtsp package. I notice a high consumption of cpu. Is there a way to consume less CPU?

mpromonet, 2020-01-12:
Dear nokiovo79 
Reducing the resolution will obviously help.
Feel free to submit PR with your improvment.
Best Regards,
Michel.

nokiovo79, 2020-01-12:
Try it, but the cpu load increase for each video source. I think that send a rtsp packet, instead of frame is more efficiently

mpromonet, 2020-01-12:
Dear nokiovo79
Did you look to [janus-gateway](https://github.com/meetecho/janus-gateway) ? It relays the RTP packets from an RTSP connection to WebRTC.
Best Regards,
Michel.

allenzhuyi, 2020-04-17:
I also test and found cpu is very high , with the increase of rtsp url num, how to reduce cpu rate?

mpromonet, 2020-04-17:
Dear allenzhuyi,

Increasing the number of rtsp url configured should not have impact, only when the stream is asked the rtsp backend is connected, and then use CPU to uncompress the stream.   
Do not hesitate to share your improvment.

Best Regards,
Michel.

allenzhuyi, 2020-04-17:
Thanks for reply. I tested with embed page. When I play one rtsp camera,the cpu rate is about 80% , when I play both two rtsp url on page, the cpu is 190%. My server is virtual machine on vmware esxi. The Hardware config is 2*4 cpu,8GB mem. How can reduce the cpu rate? Thanks.

neuromozg, 2020-04-21:
Hello! Join the comment above. Ubuntu 18.04 is installed on a VM (4 CPU, 8Gb), and v4l2rtspserver 4 cameras 800x600x20 is connected to the webrtc-streamer. When a page is viewed by multiple people, CPU performance can reach 200% or higher. What can I try to do in this situation? Thanks.

mpromonet, 2020-04-21:
Hi,

I am sorry if I not really answer to this question that was debated many times.

In order to reduce resource usage :
- reduce the resolution and the framerate,
- use the parameter `bitrate` it is possible to reduce the output bitrate.

Others ways could be to use more hardware compression/decompression (may depends on hardware and WebRTC compilation flags) or use some solution like janus-gateway that doesnot uncompress (and then doesnot adapt to client & bandwidth capacities)

Best Regards,
Michel.

neuromozg, 2020-04-21:
Thanks for the answer! You can then describe your system configuration, which does not have such problems. And webrtc build parameters. I don't care what I put on the VM. Thanks!

mpromonet, 2021-03-07:
Hi nokiovo79,

Implementing NullDecoder & NullEncoder seems a possible way to reduce CPU consuption, it's really early testing.

You may try to run webrtc-streamer with `-o` option, this will store RTP H264 payload frame encoded and forward it to WebRTC. It use less ressource but   is less adaptative (no resize, no bandwidth adaptation, no codec adaptation).

Best Regards,
Michel.

mpromonet, 2021-03-14:
Hi,

I am closing this issuer as new `-o` option give a possible way avoiding decoding/encoding.

Best Regards,
Michel.

closed 2021-03-14 by mpromonet