mpromonet/webrtc-streamer #301: Streaming live CCTV feed to Android / iOS App

guruvishnuvardan, 2020-02-04:
Hi,
**It's a question.
Can you please let me know, if I can integrate RTSP live streaming to an Android / iOS app?? , If yes, Can you please let me know the process or links to learn.

Thanks
Guru

mpromonet, 2020-02-06:
Hi Guru,
This project makes a gateway rtsp to webrtc, then connecting to the webserver on android/ios should show you the webrtc streams.
You can try with the demo app https://webrtc-streamer.herokuapp.com/
Using Chrome on android it is working, I have no IOS maybe it is working, maybe it needs some debug/modification...
Best Regards,
Michel.

vegalou, 2020-02-07:
**By my test logs,**

### 1. RTSP format

If you want to _`forward RTSP to mobile`_, push RTSP source to cloud RTSP server, then receive streaming in APP directly. Or if there is TUTK($) like P2P toolkit between IPCam and mobile, they could connect through STUN directly.

### 2. standard html5 video with webrtc

If you're running webrtc-streamer, read the README.MD homepage shows, poll the RTSP streaming via webrtc-streamer, it's HTML5 VIDEO in webrtc already, If it's behind a firewall, then push your source to gateway like Janus via webpage or nodejs cli resource.

Then all pc/mobile browsers could watch HTML5 VIDEO directly.

closed 2020-04-15 by mpromonet