mpromonet/webrtc-streamer #302: How to deploy the streamer on WAN

laowangsir, 2020-02-09:
If I want clients on internet to visit the video stream  （via UDP）, should I run a ice server ?   ( In our country , google server is blocked)

vegalou, 2020-02-11:
Webrtc-streamer runs webrtc p2p html5 sharing successfully with Janus or other webrtc gateway as README.md first page describes. Default STUN service based on Google.

#### NOT AVALIABLE TO GOOGLE CONNECTION?
If ISP blocked to google turn server like `CN` or `KR`, then there is necessary to setup COTURN your server, and specify ID/PASS to it. *Or you just belive your TURN security.*

mpromonet, 2020-02-13:
Hi @laowangsir,

I think a simple way could be to start webrtc-streamer with `-S` option, this start an embeded STUN server, that will be used by the html pages (getting it from /api/getIceServers).

Best Regards,
Michel.

closed 2020-04-26 by mpromonet