mpromonet/webrtc-streamer #303: GLIBCXX_3.4.26 not found on ubuntu 18.10

D1plo1d, 2020-02-12:
**Describe the bug**
web-streamer  0.2.3 is unable to start due to the following error on ubuntu 18.10 (x86-64):

```
./webrtc-streamer 
./webrtc-streamer: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by ./webrtc-streamer)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install web-streamer 0.2.3 from releases on ubuntu 18.10
2. run ./webrtc-streamer

**Expected behavior**
Webrtc streamer should start without error

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.10

D1plo1d, 2020-02-12:
It appears a similar issue was marked as resolved (#258) so perhaps this is a regression or a Ubuntu 18.10-specific issue?

D1plo1d, 2020-02-12:
Installing the snap resolved the issue for me (`sudo snap install webrtc-stream`).

mpromonet, 2020-02-13:
Hi @D1plo1d,
The fix #258 was does only for arm, I need to report it on x86_64.
Best Regards,
Michel.

closed 2020-02-13 by mpromonet

mpromonet, 2020-02-14:
Hi Rob,
This is fixed in https://github.com/mpromonet/webrtc-streamer/releases/download/v0.2.4/webrtc-streamer-v0.2.4-Linux-x86_64-Release.tar.gz
Best Regards,
Michel.

D1plo1d, 2020-02-15:
That fixed it for me. Thanks Michel.

commit 4aa5080b (2020-02-13): fix #303: static link stdc++ lib for all linux