mpromonet/webrtc-streamer #304: Is two-way audio supported?

mutasim, 2020-02-14:
**Use case**: I have an ONVIF compliant camera, which supports RTSP backchannel audio. I want to build a browser based, audio-only intercom for receiving but also sending back audio over WebRTC. 

Does this project support the sending of WebRTC audio back to the RTSP feed?

mpromonet, 2020-02-14:
Hi mutasim, 
This is actually not implemented, but the audio can be sent to server using :
`webRtcServer.connect(url.video,url.audio,options,stream)`
Next you should find a way to forward to the RTSP consumer, and then make stuff in live555 to send audio through the RTSPClient object.
Best Regards,
Michel.

closed 2020-10-09 by mpromonet

AleXoundOS, 2021-08-23:
@mutasim, have you succeeded?

mutasim, 2021-08-23:
No sorry - no progress from me.