mpromonet/webrtc-streamer #305: Memory optimization

rahul3349, 2020-02-24:
I'm using webrtc-streamer on linux platform, I can see that RAM usage is up to 4GB, so is there any possiblity of reducing memory foot print?

rahul3349, 2020-02-24:
I checked webrtc memory consuption by running video loopback test app which also do transcoding, it consumes ~950MB. so Need to know where the rest of memory (~3GB) is being consumed?

rahul3349, 2020-02-24:
Measured memory usage by live555 client by running openRTSP test app, it is comsuming ~60MB.

mpromonet, 2020-02-26:
Hi,
Reducing the resolution, the number of connection should help.
Feel free to submit pull request with your improvement.
Best Regards,
Michel.

closed 2020-02-26 by mpromonet

rahul3349, 2020-02-26:
@mpromonet  I'm checking memory usage without any connection.