mpromonet/webrtc-streamer #307: How do I specify the resolution of a v4l2 device?

shepmaster, 2020-03-09:
**Describe the bug**

I have a Raspberry Pi Zero W with a camera attached. This camera is a V4L2 device:

```
$ v4l2-ctl -d /dev/video0 --info
Driver Info:
	Driver name      : bm2835 mmal
	Card type        : mmal service 16.1
	Bus info         : platform:bcm2835-v4l2
	Driver version   : 4.19.97
	Capabilities     : 0x85200005
		Video Capture
		Video Overlay
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x05200005
		Video Capture
		Video Overlay
		Read/Write
		Streaming
		Extended Pix Format

$ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YU12' (Planar YUV 4:2:0)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[2]: 'RGB3' (24-bit RGB 8-8-8)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[3]: 'JPEG' (JFIF JPEG, compressed)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[4]: 'H264' (H.264, compressed)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[5]: 'MJPG' (Motion-JPEG, compressed)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[6]: 'YVYU' (YVYU 4:2:2)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[7]: 'VYUY' (VYUY 4:2:2)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[8]: 'UYVY' (UYVY 4:2:2)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[9]: 'NV12' (Y/CbCr 4:2:0)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[10]: 'BGR3' (24-bit BGR 8-8-8)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[11]: 'YV12' (Planar YVU 4:2:0)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[12]: 'NV21' (Y/CrCb 4:2:0)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
	[13]: 'BGR4' (32-bit BGRA/X 8-8-8-8)
		Size: Stepwise 32x32 - 2592x1944 with step 2/2
```

This device is detected by webrtc-streamer, but only as a tiny postage stamp in terms of resolution.

**To Reproduce**
- Unclear

**Expected behavior**

I'd hope that I'd be able to get a resolution of 1280x720 (or even 1920x1080).

I do not see any way of passing configuration options to webrtc-streamer, and it appears to override any that I've set with `v4l2-ctl` before starting.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser: Chrome/Firefox 
 - Version: Version 82.0.4078.0 (Official Build) canary (64-bit) / 73.0.1 (64-bit)

**Additional context**
N/A

mpromonet, 2020-03-09:
Hi Jake,

You could ask for a resolution of V4L2 device setting options `width` and `height` in the `/call` API.
This should be possible appending to the url  `?width=1280&height=720`

Best Regards,
Michel.

shepmaster, 2020-03-09:
Thanks! I have a feeling I'm missing some nuance, however.

I added those options to the URL: 

```
http://192.168.3.126:8000/webrtcstreamer.html?video=mmal%20service%2016.1&options=rtptransport%3Dtcp%26timeout%3D60&width=1280&height=720
```

I see the request to `/call` does not include them:

```
http://192.168.3.126:8000/api/call?peerid=0.7742015006191183&url=mmal%20service%2016.1&options=rtptransport%3Dtcp%26timeout%3D60
```

I am running WebRTC-Streamer v0.2.4/Linux-armv6l; is this an option that has not been released yet?

mpromonet, 2020-03-11:
Hi,
You should encode the `&` and `=` in the value of options argument.
Something like : 
`http://192.168.3.126:8000/webrtcstreamer.html?video=mmal%20service%2016.1&options=rtptransport%3Dtcp%26timeout%3D60%26width%3D1280%26height%3D720`
Best Regards,
Michel.

bremensaki, 2020-03-18:
I'm having a similar issue and I'm not sure how to specify the capture resolution. The suggestion of adding the width and height options as per the last comment doesn't really help, as it doesn't change the source resolution, it just scales it up.

I'm testing on an old HP laptop with integrated webcam. I've directly captured from it using ffmpeg at 640x480 with no issues, v4l2-ctl reports 640x480@30fps is a valid option, but the webrtc-streamer feed always comes in at 320x240.

mpromonet, 2020-03-18:
Hi @bremensaki,

This depends on the first connection to the v4l2 device you asked.
Next WebRTC strems are only scaled, as v4l2 device can serve only one resolution.

Did you try to close all WebRTC streams (you may close them using /admin.html page) and next ask for the first WebRTC stream request specifying width & height ?

An other way is to modify the webrtcstreamer.json.

Best Regards,
Michel.

shepmaster, 2020-03-19:
Thank you. I was confused by your original suggestion:

> This should be possible appending to the url `?width=1280&height=720`

But your follow up clarified that it wasn't appended to the URL, but instead embedded in the `options` parameter:

> in the value of options argument.

Unfortunately, setting the size to anything reasonable did not work; it simply used too much CPU.

As you mentioned in other issues, I've switched away from webrtc-streamer to using ffmpeg to directly copy the x264 stream from my v4l2 device to Janus, which then directly copies it out to clients. I also directly copy the stream to homebridge for use with homekit. This results in negligible CPU usage (~10% CPU on a Raspberry Pi Zero W). Amusingly, the most complex CPU part is creating snapshots of the stream every few seconds for homebridge! This is because that actually has to process the video stream.

Thank you for your project, and I wish you the best of luck with it!

mpromonet, 2020-03-19:
Hi shepmaster,

For a low CPU consumption, the janus solution is good, the drawback is that web browser need to support the camera codec and no bandwidth adaptation is possible (like down scaling or reduce framerate).

Maybe it was improved, but last time I looked to the web integration was quite complex, making a page with video room is not trivial.

Best Regards,
Michel.

closed 2020-04-26 by mpromonet