mpromonet/webrtc-streamer #309: How is free public access with webrtc-streamer.herokuapp.com possible?

generic-beat-detector, 2020-03-17:
Hello mpromonet,

Firstly, allow me to express my appreciation for your wonderful project! 

I have an old i5-3230M quad core (@ 2.60GHz per core) hosting `webrtc-streamer` and serving as an edge gateway for my modest flat. I've installed RPi's around the place and can access the streams from practically anywhere. 

Now, I noticed that opening a single stream resulted in about 30% usage on a single core, 2 streams doubled that, so on and so forth -- well at least according to `top(1)`. So, the big mystery is how your free demo via `webrtc-streamer.herokuapp.com` is able to sustain (1000's? of) random and simultaneous requests so smoothly? Something has to give:

* Is the herokuapp merely a signaling utility -- with the locations on the demo webpage performing the actual rtsp to webrtc and direct STUN P2P media streaming with each random connection (and thus bearing the CPU load)? 

*  Or is herokuapp the intermediary rtsp to webrtc server, handing all media conversions and streaming -- in which case maintaining it must cost you a non-trivial amount?

Regards.

mpromonet, 2020-03-17:
Hi GDB,
Probably the resolution have a big impact on the needed ressources.
I am using the free plan of heroku that allow to run full time 1 app.
It seems something like a 64GB 8core. 
The backend stream are shared, but I don't think it could handle 1000 streams.
Best Regards,
Michel.

generic-beat-detector, 2020-03-17:
Nice -- and thank you for the prompt reply. 

Just to clarify one little detail: when you state "The backend stream[s] are shared", you imply that the PaaS will continue to indiscriminately serve random (WebRTC-side) connections until "it can take no more?". Please forgive my dull, dim-witted mind -- I'm just slow like that...

mpromonet, 2020-03-18:
Hi,

My meaning by shared, is that when a WebRTC stream is asked (calling /call api) it connects to the RTSP server, if the same stream is asked, the RTSP consumer is connected to these 2 WebRTC producers. When there is no more WebRTC consumer for an RTSP stream, the RTSP stream is closed.

I think if you connected lots of RTSP consumer and WebRTC producer, it probably slow down or crashed depending if it run out of memory or out of CPU.

Best Regards,
Michel.

generic-beat-detector, 2020-03-18:
Precision!

closed 2020-03-18 by generic-beat-detector