mpromonet/webrtc-streamer #310: New Firefox 74.0 Feature Affecting WebRTC Compatibility?

generic-beat-detector, 2020-03-18:
Hello mpromonet (and all),

__Firefox Issue__

About two days ago, on my Ubuntu 18.04.2 LTS laptop, Firefox upgraded itself to release 74.0 (by "asking" me to restart it). After this upgrade, video streams from my local docker installation of `webrtc-streamer` __v0.2.3-19-g49b8daf/Linux-x86_64__ (on a separate machine) no longer appeared on the dark grey HTML5 video element -- which turns light grey after a while.

I pulled the latest `webrtc-streamer` docker image, __v0.2.4-52-g7aa1239/Linux-x86_64__, and restarted the service (on the separate machine), but still no video streams appeared on Firefox 74.0 (on my laptop). Yet, a very ancient Android phone running Firefox 68.3.0 still showed the video streams -- with both versions of `webrtc-streamer`.
 
__Firefox 74.0 Quick Fix (Not Recommended)__

According to [Kinda urgent, firefox no longer compatible (software.intel.com)](https://software.intel.com/en-us/forums/intel-collaboration-suite-for-webrtc/topic/850221), apparently a new Firefox 74.0 feature that obfuscates the "computer’s IP address with a random ID in certain WebRTC scenarios" is the likely culprit.

Navigating to the `about:config` URL on Firefox 74.0 and toggling `media.peerconnection.ice.obfuscate_host_addresses` to `false` seemingly does the trick. The `webrtc-streamer` video now appears on the Firefox 74.0 HTML5 video elements.

Regards.

mpromonet, 2020-03-19:
Hi generic-beat-detector,

I made a test with the live demo on http://webrtc-streamer.herokuapp.com/ with firefox 74, it seems to work normally.

So maybe the problem you are facing is link to how you are running docker.
Did you try adding `--net=host`  ?

Best Regards,
Michel.

generic-beat-detector, 2020-03-19:
Yes, Michel, yes. 

Adding the `--network host` switch fixed the problem. 

I restored `media.peerconnection.ice.obfuscate_host_addresses` to `true` and video streaming still worked. If you ever get the time, it will be quite a pleasure to hear the logic behind this fix.

__For what its worth:__

I tried to obtain logs for both scenarios so that I could run a diff and see if I could maybe identify the problem. So, with `media.peerconnection.ice.obfuscate_host_addresses` set to `true`:

* I ran docker without the `--network host` switch, connected with FF 74.0, and (as expected) no video output
* Brought up the browser's Web Console (`CTRL+SHIFT+K`) and selected the `Network` tab, and clicked on the `Reload` button in the

    "Perform a request or `Reload` the page to see detailed information about network activity." 

    option. The page reloaded and this time the video stream appeared! 
* Closing the Web Console and reloading (`CTRL+R`) ... no video
* Leaving the Web Console open and reloading (`CTRL+R`) ... video!?

Whatever FF, I'm sticking with Michel's `--network host` fail-safe.

Best Regards.

mpromonet, 2020-03-21:
Hi,

This is something into I need to dig a bit much, but the idea behind is that the ICE candidates expose internal docker IP. Using  `--network host` makes the docker to see your host ip address that will be published as an ICE candidate.

Best Regards,
Michel.

generic-beat-detector, 2020-03-21:
Oh, without `--network host`, its as if we have a "virtually NATed" `webrtc-stremer` with no "STUN", right?

s-beard, 2020-04-16:
Hi everybody,

I have Exactly the same problem, but my application don't run with docker.
I have compile the webrtc-streamer on debian 9, all is working when i call my rtsp video on chrome and firefox 73.
But on firefox 75 my test is KO.
When i toggle the `media.peerconnection.ice.obfuscate_host_addresses` option to false it works fine.

I run the webrtc-streamer like this : `/webrtc-streamer -H8000 -S`

Can someone have a solution ?

mpromonet, 2020-04-16:
Hi s-beard,

I just made a try with firefox 75 on windows without overiding media.peerconnection.ice.obfuscate_host_addresses and it works well.    
Do you have a problem with test application on https://webrtc-streamer.herokuapp.com/Bunny.html ?

Best Regards,
Michel.

s-beard, 2020-04-17:
Hi Michel,
No i don't have any problems with your Bunny video. 
I run the webrtc-streamer like this : `/webrtc-streamer -H8000 -S` for watching an rtsp video live, or rtsp video replay from hikvision recorder.

[webrtc-streamer.log](https://github.com/mpromonet/webrtc-streamer/files/4493444/webrtc-streamer.log)
[Firefox.log](https://github.com/mpromonet/webrtc-streamer/files/4493452/Firefox.log)

Best Regards,

closed 2020-07-04 by mpromonet