mpromonet/webrtc-streamer #312: [Usage] https server

antinucleon, 2020-03-30:
The help info for command args "-c" is not very clear, I tried a few combinations and always get error:

```
ssl_use_pem_file: cannot open certificate file ./ssl/key.pem: error:0906D06C:PEM routines:PEM_read_bio:no start line
```

What is the correct way to start an https server?

Thanks,

mpromonet, 2020-03-30:
Hi @antinucleon 

The file should contains the private key and the certificate in PEM format.
You should get informations from https://github.com/civetweb/civetweb/blob/master/docs/OpenSSL.md

Best Regards,
Michel.

antinucleon, 2020-03-30:
Thanks! I should cat key and pem files into one file.

closed 2020-03-30 by antinucleon

yacaeh, 2020-04-08:
Hi I'm also trying to start https server, merged multiple cert files into one file in pem format
but it doesn't give any messages and when I try to allocate port to https, it doesn't work with -H option. Can you help? 

`./webrtc-streamer.exe -H 1443s -c ../cert/localhost_onefile.pem`  // seems it can't recognize https port

`./webrtc-streamer.exe -H 1443 -c ../cert/localhost_onefile.pem`  // this one is just http server but not working

`./webrtc-streamer.exe -H 0.0.0.0:1443 -c ../cert/localhost_onefile.pem`  // doesn't work