mpromonet/webrtc-streamer #313: arm version can not run.

zsinba, 2020-04-10:
**Describe the bug**
A clear and concise description of what the bug is.
I can't run it in the openwrt. it is a arm7 (ARMv7 Processor rev 5 (v7l) x 4)
I got the response when run it:
root@OpenWrt:~/webrtc-streamer-v0.2.6-Linux-armv7l-Release# ./webrtc-streamer 
-ash: ./webrtc-streamer: not foun
--

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]  
Openwrt
Arm7
OpenWrt R9.12.03 / LuCI Master (git-19.338.43082-9e87e69)
--
 - OS: [e.g. iOS8.1]
arm,Linux
 - Browser [e.g. stock browser, safari]
no
 - Version [e.g. 22]
Latest

**Additional context**
Add any other context about the problem here.

mpromonet, 2020-04-10:
Hi,
The armv7 binary is compiled with https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/Buster/GCC%209.2.0/Raspberry%20Pi%202%2C%203/cross-gcc-9.2.0-pi_2-3.tar.gz toolchain.
It produce an executable that can run on raspberry pi and nano pi, maybe others platform that are binary compatible.   
You probably need to compile with the toolchain of your board.

Best Regards,
Michel.

zsinba, 2020-04-10:
root@OpenWrt:~/webrtcstreamer# ldd webrtc-streamer 
	/lib/ld-linux-armhf.so.3 (0xb6eba000)
	libdl.so.2 => /lib/ld-linux-armhf.so.3 (0xb6eba000)
	libm.so.6 => /lib/ld-linux-armhf.so.3 (0xb6eba000)
	libpthread.so.0 => /lib/ld-linux-armhf.so.3 (0xb6eba000)
	libc.so.6 => /lib/ld-linux-armhf.so.3 (0xb6eba000)
Error loading shared library ld-linux-armhf.so.3: No such file or directory (needed by webrtc-streamer)
Error relocating webrtc-streamer: __memcpy_chk: symbol not found
Error relocating webrtc-streamer: __longjmp_chk: symbol not found
Error relocating webrtc-streamer: __isinf: symbol not found
Error relocating webrtc-streamer: __vsnprintf_chk: symbol not found
Error relocating webrtc-streamer: __fprintf_chk: symbol not found
Error relocating webrtc-streamer: __strftime_l: symbol not found
Error relocating webrtc-streamer: __fdelt_chk: symbol not found
Error relocating webrtc-streamer: __isnan: symbol not found
Error relocating webrtc-streamer: fcntl64: symbol not found
Error relocating webrtc-streamer: __sprintf_chk: symbol not found
root@OpenWrt:~/webrtcstreamer# ./webrtc-streamer 
-ash: ./webrtc-streamer: not fo

zsinba, 2020-04-10:
coud you please support openwrt verson?

mpromonet, 2020-04-10:
The loader seems different from the raspberry one or the libc is not the same.

Did you tried to build with the toolchain corresponding to your plateform ?

zsinba, 2020-04-10:
I still don't know how to compile. I am still looking at how to compile into openwrt.
I'm still looking for information, can you help?

mpromonet, 2020-04-10:
Hi,

I guess you have 2 options either build an executable using openwrt build mechanism or build an executable in another way that is compatible with the openwrt runtime.
You may find information about openwrt https://openwrt.org/docs/guide-developer/packages 

Best Regards,
Michel.

zsinba, 2020-04-11:
谢谢，我试一下。

zsinba, 2020-04-13:
然而并没有成功。
希望作者能出Openwrt版本的。

closed 2025-01-18 by mpromonet