mpromonet/webrtc-streamer #315: Window Capturing Issues

nagarjunabandi, 2020-04-17:
I have integrated screen capturer module in webrtc native peer connection example. It is working fine when selecting screen through ScreenCapturer class. Able to capture window also with WindowCapturer class only when the window is in maximized state. If the window restored or resized then frames are getting disturbed. Also unable to capture cursor in ScreenCapturer mode as well as WindowCapturer mode.

[image]

mpromonet, 2020-04-17:
Hi nagarjunabandi,

Yes I know, on Windows it works with screen, but not with windows... with older WebRTC SDK it was working. It works on Linux.
Probably a bas computation of stride... if you find the problem thanks to share it.

Best Regards,
MicheL

nagarjunabandi, 2020-04-21:
Thanks for your quick reply, I will look into the window capture issue, can you please help me out to capture mouse cursor. Thanks in advance.

commit 05ff828f (2020-09-22): fix #315: on windows desttop frame stride is not equal to width*4
closed 2020-09-22 by mpromonet