mpromonet/webrtc-streamer #316: DEMO-APP and Raspberry Stream not accessible

beniroquai, 2020-04-19:
**Describe the bug**
When accessing the webpage ```https://webrtc-streamer.herokuapp.com/```, the streams are not visible from a german IP-address. When switching to an american using (Anyconnect) VPN, you can see them clearly. 

When trying to access a video stream from the raspberry pi with internal and external STUN-server the webpage renders on a remote machine, but the stream stays gray.

Any idea where the problem could be found? All necessary ports (8000, TCP/UPD 3478) are opened. 

I have a public IPv4 IP. No IPv6 available. 

**To Reproduce**
Steps to reproduce the behavior:

I start the server with ```./webrtc-streamer -H 0.0.0.0:8000 -sstun4.l.google.com:19302```

**Desktop (please complete the following information):**
 - OS: Raspbian (ltest)
 - Browser: Chrome, Firefox (latest

mpromonet, 2020-04-19:
Hi Benedict,

I am not sure to understand, the demo app select ramdomly the webrtc stream, some of the RTSP url works or not it depends. When the backend doesnot send images, the image stray grey.  
In many case the Bunny streams is working, do you have the same behaviour using http://webrtc-streamer.herokuapp.com/?video=Bunny ?

Best Regards,
Michel.

beniroquai, 2020-04-19:
Interesting the Bunny movie works on my local network where the Pi is treaming to the public, but not on my remote computer: 
[image]

and cellphone:
[image]

Any ideas what could be the bottleneck here?

beniroquai, 2020-04-19:
The video with WiFi enabled on phone works.

mpromonet, 2020-04-19:
@beniroquai 

When the browser & webrtc-streamer are on same network, it doesnot need STUN, the ICE candidate with the local IP is ok to communicate.

You may try the anwser https://stackoverflow.com/a/35437308/3102264 to see if you can reach the STUN server stun:stun.l.google.com:19302 used by the demo app on heroku.

Best Regards,
Michel.

beniroquai, 2020-04-19:
Hey @mpromonet, thanks a lot for the quick reply! I've tried the code with a set of different STUNs and it seems that it cannot connect to any of the servers at all. At least it does not come any further than that: 

```
BASH$ ./stun 62.71.2.168  19302 8888
Main start ... 
socket opened to  62.71.2.168:19302  at local port 8888
Send data ...
Read recv ...
```

And of course you're right, inside the local network it works without the STUN :)

mpromonet, 2020-04-19:
Hi @beniroquai ,

Strange DNS give me stun.l.google.com 66.102.1.127
With this IP,I can connect to STUN, but with 62.71.2.168, it fails too from France.

Best Regards,
Michel.

beniroquai, 2020-04-19:
Sorry, I posted the result form a different STUN  (stun.rixtelecom.se), since ``` stun.l.google.com``` gives me this output:
`BASH$ nslookup stun.l.google.com:19302 
Server:		192.168.2.1
Address:	192.168.2.1#53

** server can't find stun.l.google.com:19302: NXDOMAIN
`

beniroquai, 2020-04-19:
Ah, sorry, I made a mistake, I forgot to delete the PORT! Now I get this:

```
BASH$ ./stun 74.125.133.127  19302 8000
Main start ... 
socket opened to  74.125.133.127:19302  at local port 8000
Send data ...
Read recv ...
STUN binding resp: success !
socket closed !
ip:port = 93.205.114.61:8000
Main over.
```

So, it sees the STUN it seems.

beniroquai, 2020-04-19:
Same situation with my cellphone being a hotspot for my laptop (where the above message came from - and the stream actually works,, also from WAN-IP):
```
BASH$ ./stun 74.125.133.127  19302 8000
Main start ... 
socket opened to  74.125.133.127:19302  at local port 8000
Send data ...
sendto error
STUN req error : -1
Main over.
```

So this means, that the STUN is not reachable?

beniroquai, 2020-04-20:
Ok, next thing which is confusing. I have started the STUN-server in webrtc-streamer with this command:
``` ./webrtc-streamer -S -s$(curl ifconfig.me -s):3478 ```

and get the following output on the computer where I cannot see the BUnny stream from the DEMO app:
```
BASH$ ./stun  X.X.X.X 3478  8000
Main start ... 
socket opened to  X.X.X.X:3478  at local port 8000
Send data ...
Read recv ...
```
where X.X.X.X is my global IP Adress visible from the outside world. 
The same behaviour is reproducible with the package ```coturn```.  Could that mean, that the port is simply blocked by the internet provider?

beniroquai, 2020-04-20:
Further, ```coturn``` gives me this:

```

0: SQLite DB connection success: /var/lib/turn/turndb
5: handle_udp_packet: New UDP endpoint: local addr 192.168.2.118:8888, remote addr X.X.X.X:8000
5: session 000000000000000001: realm <youseetoo.ddns.net> user <>: incoming packet BINDING processed, success
```

which means it connects to the server, but the ```./stun``` program doesn't receive anything back from the server. I'm puzzled...

beniroquai, 2020-04-20:
Some clarificatoin. I managed to get the STUN/TURN-server on the raspberry pi working now. It's now visible from remote clients.

The following error in webrtc-streamer is given once I try to connect to it trough the website from a remote client:

```

BASH:~/Downloads/webrtc-streamer-v0.2.6-11-gbca3ae0-Linux-armv7l-Release $ ./webrtc-streamer  -sX.X.X.X:3478
Version:v0.2.6-11-gbca3ae0/Linux-armv7l civetweb@v1.11 webrtc@29cf2b0cbb-dirty live555helper@b4d86f2
nullLogger level:3
HTTP Listen at 0.0.0.0:8000
uri:uri:/api/getIceServers/api/version

answer:"v0.2.6-11-gbca3ae0/Linux-armv7l civetweb@v1.11 webrtc@29cf2b0cbb-dirty live555helper@b4d86f2"
answer:{
        "iceServers" :
        [
                {
                        "urls" :
                        [
                                "stun:X.X.X.X:3478"
                        ]
                }
        ]
}
uri:/api/call
[000:000] [12526] (PeerConnectionManager.h:223): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.20278384884286083
[000:015] [12526] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.20278384884286083
[000:023] [12526] (webrtc_video_engine.cc:3251): Absent receive stream; ignoring clearing encoded frame sink for ssrc 0
[000:024] [12526] (PeerConnectionManager.h:208): virtual void PeerConnectionManager::PeerConnectionObserver::OnAddStream(rtc::scoped_refptr<webrtc::MediaStreamInterface>) nb video tracks:1
[000:416] [12530] (CapturerFactory.h:195): audiourl:mmal service 16.1 idx_audioDevice:-1
[000:416] [12530] (PeerConnectionManager.cpp:1113): Cannot create capturer audio:
[000:417] [12526] (PeerConnectionManager.h:223): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.20278384884286083
[000:426] [12526] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.20278384884286083
answer:{
        "sdp" : "v=0\r\no=- 1364528249479174857 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE 0 1 2\r\na=msid-semantic: WMS mmalservice161||\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 9 0 8 105 13 110 113 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:cdyi\r\na=ice-pwd:6GFxk7ITp42LBTgXe9UknnXC\r\na=ice-options:trickle\r\na=fingerprint:sha-256 AF:96:25:54:AE:22:C3:F7:EB:2F:C3:CE:5F:7D:0C:B9:9E:14:B9:68:8F:64:42:BE:B6:6B:EA:B2:4C:86:C6:A6\r\na=setup:active\r\na=mid:0\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:3 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 97 98 99 102 122 127 121 125 107 108 109 114 115 116\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:cdyi\r\na=ice-pwd:6GFxk7ITp42LBTgXe9UknnXC\r\na=ice-options:trickle\r\na=fingerprint:sha-256 AF:96:25:54:AE:22:C3:F7:EB:2F:C3:CE:5F:7D:0C:B9:9E:14:B9:68:8F:64:42:BE:B6:6B:EA:B2:4C:86:C6:A6\r\na=setup:active\r\na=mid:1\r\na=extmap:14 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:2 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:13 urn:3gpp:video-orientation\r\na=extmap:3 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:12 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=extmap:11 http://www.webrtc.org/experiments/rtp-hdrext/video-content-type\r\na=extmap:7 http://www.webrtc.org/experiments/rtp-hdrext/video-timing\r\na=extmap:8 http://tools.ietf.org/html/draft-ietf-avtext-framemarking-07\r\na=extmap:9 http://www.webrtc.org/experiments/rtp-hdrext/color-space\r\na=sendrecv\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:96 VP8/90000\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:98 VP9/90000\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=fmtp:98 profile-id=0\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:102 H264/90000\r\na=rtcp-fb:102 goog-remb\r\na=rtcp-fb:102 transport-cc\r\na=rtcp-fb:102 ccm fir\r\na=rtcp-fb:102 nack\r\na=rtcp-fb:102 nack pli\r\na=fmtp:102 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42001f\r\na=rtpmap:122 rtx/90000\r\na=fmtp:122 apt=102\r\na=rtpmap:127 H264/90000\r\na=rtcp-fb:127 goog-remb\r\na=rtcp-fb:127 transport-cc\r\na=rtcp-fb:127 ccm fir\r\na=rtcp-fb:127 nack\r\na=rtcp-fb:127 nack pli\r\na=fmtp:127 level-asymmetry-allowed=1;packetization-mode=0;profile-level-id=42001f\r\na=rtpmap:121 rtx/90000\r\na=fmtp:121 apt=127\r\na=rtpmap:125 H264/90000\r\na=rtcp-fb:125 goog-remb\r\na=rtcp-fb:125 transport-cc\r\na=rtcp-fb:125 ccm fir\r\na=rtcp-fb:125 nack\r\na=rtcp-fb:125 nack pli\r\na=fmtp:125 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na [... cut ...]
                "candidate" : "candidate:1152796752 1 udp 1686052607 X.X.X.X 59562 typ srflx raddr 192.168.2.118 rport 59562 generation 0 ufrag cdyi network-id 1",
                "sdpMLineIndex" : 0,
                "sdpMid" : "0"
        },
        {
                "candidate" : "candidate:2340270612 1 tcp 1518280447 192.168.2.118 46207 typ host tcptype passive generation 0 ufrag cdyi network-id 1",
                "sdpMLineIndex" : 0,
                "sdpMid" : "0"
        }
]
answer:true
```

So it seems everything is fine. Hm.

herpiko, 2020-04-23:
The bunny is still not accessible not all.

https://webrtc-streamer.herokuapp.com/Bunny.html

[image]

Log, https://gist.githubusercontent.com/herpiko/ddc924a4cafe13ac3e2ac2064902df34/raw/aca4e2b44313e90d59a77b7351080ba9853d2b99/gistfile1.txt

On Firefox, it shows ICE server related error,
```
addIceCandidate OK 2 webrtcstreamer.js:286:33
oniceconnectionstatechange  state: checking webrtcstreamer.js:160:11
addIceCandidate OK webrtcstreamer.js:286:33
addIceCandidate ok:true 2 webrtcstreamer.js:227:13
ICE failed, add a TURN server and see about:webrtc for more details
oniceconnectionstatechange  state: failed
```

Wallacy, 2020-10-22:
Same here FWIW:
OSX / Chrome.

closed 2021-01-20 by mpromonet