mpromonet/webrtc-streamer #317: I am facing GLIBC version issue whiling creating docker based on ubuntu 16.04

harendra247, 2020-04-30:
**Describe the bug**
I want to run webrtc-steamer binary in ubuntu 16.04 based docker container. but I am facing GLIBC version issue. GLIBC_2.23 is available on ubuntu 16.04, but the binary needs GLIBC_2.27. 

I even tried by making binary based on FROM heroku/heroku:16 but I faced compilation issue.

```
./webrtc-streamer: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.27' not found (required by ./webrtc-streamer)
The command '/bin/sh -c apt-get update && apt-get install -y --no-install-recommends libasound2 libgtk-3-0 libssl1.0    && apt-get clean && rm -rf /var/lib/apt/lists/     && ./webrtc-streamer -V' returned a non-zero code: 1

```

**To Reproduce**
Steps to reproduce the behavior:
1. Open the Dockerfile 
2. Change 'FROM ubuntu:18.04' to 'FROM ubuntu:16.04'

**Expected behavior**
docker should be created.

**Desktop (please complete the following information):**
 - OS: Linux 16.04 x86_64 and
- OS: Linux 16.04 arm64

Please let me know if more info is needed.

harendra247, 2020-04-30:
When I am compiling on my host directly I am facing issue 

```
FAILED: obj/api/video/video_stream_encoder_create/video_stream_encoder_create.o
../../third_party/llvm-build/Release+Asserts/bin/clang++ -MMD -MF obj/api/video/video_stream_encoder_create/video_stream_encoder_create.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_X11=1 -DGOOGLE_CHROME_BUILD -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_GNU_SOURCE -DCR_CLANG_REVISION=\"n346557-4e0d9925-2\" -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DRTC_ENABLE_VP9 -DHAVE_SCTP -DWEBRTC_USE_H264 -DWEBRTC_LIBRARY_IMPL -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -I../.. -Igen -I../../third_party/abseil-cpp -I../../third_party/libyuv/include -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -B../../third_party/binutils/Linux_x64/Release/bin -pthread -fcolor-diagnostics -fmerge-all-constants -fcrash-diagnostics-dir=../../tools/clang/crashreports -Xclang -mllvm -Xclang -instcombine-lower-dbg-declare=0 -m64 -march=x86-64 -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -Xclang -fdebug-compilation-dir -Xclang . -no-canonical-prefixes -Wall -Wextra -Wimplicit-fallthrough -Wunreachable-code -Wthread-safety -Wextra-semi -Wno-missing-field-initializers -Wno-unused-parameter -Wno-c++11-narrowing -Wno-unneeded-internal-declaration -Wno-undefined-var-template -Wno-ignored-pragma-optimize -Wno-implicit-int-float-conversion -Wno-final-dtor-non-final-class -Wno-builtin-assume-aligned-alignment -Wno-deprecated-copy -Wno-non-c-typedef-for-linkage -Wno-pointer-to-int-cast -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -ftrivial-auto-var-init=pattern -fvisibility=hidden -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -Wexit-time-destructors -Wglobal-constructors -Wc++11-narrowing -Wimplicit-fallthrough -Wthread-safety -Winconsistent-missing-override -Wundef -Wunused-lambda-capture -Wno-shorten-64-to-32 -std=c++14 -fno-exceptions -fno-rtti -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../../api/video/video_stream_encoder_create.cc -o obj/api/video/video_stream_encoder_create/video_stream_encoder_create.o
In file included from ../../api/video/video_stream_encoder_create.cc:16:
In file included from ../../video/video_stream_encoder.h:42:
In file included from ../../video/adaptation/video_stream_encoder_resource_manager.h:17:
In file included from /usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/unordered_map:47:
In file included from /usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/hashtable.h:35:
/usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/hashtable_policy.h:85:11: error: implicit instantiation of undefined template 'std::hash<webrtc::VideoAdaptationReason>'
        noexcept(declval<const _Hash&>()(declval<const _Key&>()))>
                 ^
/usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/type_traits:138:14: note: in instantiation of template class 'std::__detail::__is_noexcept_hash<webrtc::VideoAdaptationReason, std::hash<webrtc::VideoAdaptationReason>>' requested here
    : public conditional<_B1::value, _B2, _B1>::type
             ^
/usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/type_traits:148:39: note: in instantiation of template class 'std::__and_<std::__is_fast_hash<std::hash<webrtc::VideoAdaptationReason>>, std::__detail::__is_noexcept_hash<webrtc::VideoAdaptationReason, std::hash<webrtc::VideoAdaptationReason>>>' requested here
    : public integral_constant<bool, !_Pp::value> 
                                      ^
/usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/unordered_map.h:46:34: note: in instantiation of template class 'std::__not_<std::__and_<std::__is_fast_hash<std::hash<webrtc::VideoAdaptationReason>>, std::__detail::__is_noexcept_hash<webrtc::VideoAdaptationReason, std::hash<webrtc::VideoAdaptationReason>>>>' requested here
    : public conditional<_B1::value, _B2, _B1>::type
             ^
/usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/type_traits:148:39: note: in instantiation of template class 'std::__and_<std::__is_fast_hash<std::hash<webrtc::VideoAdaptationReason>>, std::__detail::__is_noexcept_hash<webrtc::VideoAdaptationReason, std::hash<webrtc::VideoAdaptationReason>>>' requested here
    : public integral_constant<bool, !_Pp::value> 
                                      ^
/usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/unordered_map.h:46:34: note: in instantiation of template class 'std::__not_<std::__and_<std::__is_fast_hash<std::hash<webrtc::VideoAdaptationReason>>, std::__detail::__is_noexcept_hash<webrtc::VideoAdaptationReason, std::hash<webrtc::VideoAdaptationReason>>>>' requested here
           typename _Tr = __umap_traits<__cache_default<_Key, _Hash>::value>>
                                        ^
/usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/unordered_map.h:100:15: note: in instantiation of default argument for '__umap_hashtable<webrtc::VideoAdaptationReason, webrtc::VideoAdaptationCounters, std::hash<webrtc::VideoAdaptationReason>, std::equal_to<webrtc::VideoAdaptationReaso [... cut ...]
        unique_ptr(unique_ptr<_Up, _Ep>&& __u) noexcept
        ^
/usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/unique_ptr.h:228:2: note: candidate template ignored: could not match 'auto_ptr' against 'unique_ptr'
        unique_ptr(auto_ptr<_Up>&& __u) noexcept; 

/usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/unique_ptr.h:169:7: note: explicit constructor is not a candidate
      unique_ptr(pointer __p) noexcept
      ^
1 warning and 3 errors generated.
```

mpromonet, 2020-04-30:
Hi Harendra,

I guess you need to upgrade, it doesnot build on 1604 since a while, and gcc5 is too old.
It works well with 1804.
Maybe runing docker image 18.04 on 16.04.

Best Regards,
Michel.

mpromonet, 2020-04-30:
@harendra247 

Building the glibc 2.27 on ubuntu16.04 or build webrtc-streamer with static libc ?

Best Regards,
Michel.

harendra247, 2020-04-30:
Yeah I can build webrtc with static libc, Please suggest the flag to use for this.

harendra247, 2020-04-30:
Tried compiling statically by commenting the lines but same issue. @mpromonet 

```
git diff 
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 92949d7..f829833 100755
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -171,14 +171,8 @@ else()
     endif()
     # static link of stdc++ if available
     include(CheckCXXCompilerFlag)
-    CHECK_CXX_COMPILER_FLAG("-static-libgcc" CXX_SUPPORTS_STATIC_LIBGCC)
-    if (CXX_SUPPORTS_STATIC_LIBGCC) 
       target_link_libraries (${PROJECT_NAME} -static-libgcc)
-    endif()
-    CHECK_CXX_COMPILER_FLAG("-static-libstdc++" CXX_SUPPORTS_STATIC_LIBSTDCPP)
-    if (CXX_SUPPORTS_STATIC_LIBSTDCPP) 
       target_link_libraries (${PROJECT_NAME} -static-libstdc++)
-    endif()
 endif()
```

Error:

```
./webrtc-streamer: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.27' not found (required by ./webrtc-streamer)
The command '/bin/sh -c apt-get update && apt-get install -y --no-install-recommends libasound2 libgtk-3-0 libssl1.0    && apt-get clean && rm -rf /var/lib/apt/lists/     && ./webrtc-streamer -V' returned a non-zero code: 1

```

mpromonet, 2020-05-01:
Hi,

The `-static` flag is probably a bit too much, maybe without desktop capture.

Best Regards,
Michel.

harendra247, 2020-05-08:
Thanks for your support @mpromonet ..Started using Ubuntu 18.04.. Closing it.

closed 2020-05-08 by harendra247