mpromonet/webrtc-streamer #318: Unable to build webrtc

miguelwon, 2020-05-01:
Hi,
I'm not able to build webrtc. I get the error I show bellow when fechting webrtc. Any idea what could be the problem? I'm running Raspbian Buster in a Raspi 4.

```
pi@raspberrypi:~/experiments/build-webrtc $ fetch --no-history webrtc

Errors:
  failed to resolve infra/3pp/tools/git/linux-armv6l@version:2.24.1.chromium.5 (line 27): no such package
  failed to resolve infra/3pp/tools/cpython/linux-armv6l@version:2.7.17.chromium.22 (line 21): no such package
  failed to resolve infra/3pp/tools/cpython3/linux-armv6l@version:3.8.0.chromium.8 (line 24): no such package
/home/pi/experiments/build-webrtc/depot_tools/bootstrap_python3: line 32: bootstrap-3.8.0.chromium.8_bin/python3/bin/python3: No such file or directory
cat: /home/pi/experiments/build-webrtc/depot_tools/python3_bin_reldir.txt: No such file or directory
[E2020-05-01T12:49:43.829365+01:00 19007 0 annotate.go:241] original error: fork/exec /home/pi/experiments/build-webrtc/depot_tools/python3: no such file or directory
[E2020-05-01T12:49:43.834321+01:00 19007 0 annotate.go:241] 
[E2020-05-01T12:49:43.834396+01:00 19007 0 annotate.go:241] goroutine 1:
[E2020-05-01T12:49:43.834437+01:00 19007 0 annotate.go:241] #0 go.chromium.org/luci/vpython/python/interpreter.go:89 - python.(*Interpreter).GetVersion()
[E2020-05-01T12:49:43.834491+01:00 19007 0 annotate.go:241] #1 go.chromium.org/luci/vpython/venv/config.go:286 - venv.(*Config).resolvePythonInterpreter()
[E2020-05-01T12:49:43.834531+01:00 19007 0 annotate.go:241]   reason: failed to determine Python version for: /home/pi/experiments/build-webrtc/depot_tools//python3
[E2020-05-01T12:49:43.834579+01:00 19007 0 annotate.go:241] 
[E2020-05-01T12:49:43.834616+01:00 19007 0 annotate.go:241] #2 go.chromium.org/luci/vpython/venv/config.go:186 - venv.(*Config).makeEnv()
[E2020-05-01T12:49:43.834708+01:00 19007 0 annotate.go:241]   reason: failed to resolve system Python interpreter
[E2020-05-01T12:49:43.834760+01:00 19007 0 annotate.go:241] 
[E2020-05-01T12:49:43.834802+01:00 19007 0 annotate.go:241] #3 go.chromium.org/luci/vpython/venv/venv.go:150 - venv.With()
[E2020-05-01T12:49:43.834840+01:00 19007 0 annotate.go:241]   reason: failed to initialize empty probe environment
[E2020-05-01T12:49:43.834884+01:00 19007 0 annotate.go:241] 
[E2020-05-01T12:49:43.834927+01:00 19007 0 annotate.go:241] #4 go.chromium.org/luci/vpython/run.go:62 - vpython.Run()
[E2020-05-01T12:49:43.834964+01:00 19007 0 annotate.go:241] #5 go.chromium.org/luci/vpython/application/application.go:320 - application.(*application).mainImpl()
[E2020-05-01T12:49:43.835010+01:00 19007 0 annotate.go:241] #6 go.chromium.org/luci/vpython/application/application.go:406 - application.(*Config).Main.func1()
[E2020-05-01T12:49:43.835056+01:00 19007 0 annotate.go:241] #7 go.chromium.org/luci/vpython/application/support.go:46 - application.run()
[E2020-05-01T12:49:43.835094+01:00 19007 0 annotate.go:241] #8 go.chromium.org/luci/vpython/application/application.go:405 - application.(*Config).Main()
[E2020-05-01T12:49:43.835130+01:00 19007 0 annotate.go:241] #9 vpython/main.go:106 - main.mainImpl()
[E2020-05-01T12:49:43.835166+01:00 19007 0 annotate.go:241] #10 vpython/main.go:112 - main.main()
[E2020-05-01T12:49:43.835202+01:00 19007 0 annotate.go:241] #11 runtime/proc.go:203 - runtime.main()
[E2020-05-01T12:49:43.835238+01:00 19007 0 annotate.go:241] #12 runtime/asm_arm.s:868 - runtime.goexit()
```

mpromonet, 2020-05-01:
Hi miguelwon,

You should cross compile, I think the WebRTC SDK deosnot build natively on arm.

Best Regards,
Michel.

miguelwon, 2020-05-01:
Oh, ok. So I need first to compile it in an external compatible environment and then use it in raspberry?

miguelwon, 2020-05-02:
So, now I get a bunch of errors at the [building](https://github.com/mpromonet/webrtc-streamer/wiki/Cross-compile-WebRTC-for-Raspberry-Pi) step with ninja. I'm running Ubuntu 16.04.

```
mwon@mwon:/mwon_disk/disk2/raspberry/build-webrtc/webrtc/src$ ninja -C arm/out/Release
ninja: Entering directory `arm/out/Release'
[1/2998] CXX obj/api/libjingle_peerconnection_api/jsep_ice_candidate.o
FAILED: obj/api/libjingle_peerconnection_api/jsep_ice_candidate.o 
arm-linux-gnueabihf-g++ -MMD -MF obj/api/libjingle_peerconnection_api/jsep_ice_candidate.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DCR_SYSROOT_HASH=3fcc1d4e44127006318371002a0f421a4fde2ab4 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DRTC_ENABLE_VP9 -DHAVE_SCTP -DWEBRTC_USE_H264 -DWEBRTC_ARCH_ARM -DWEBRTC_ARCH_ARM_V7 -DWEBRTC_HAS_NEON -DWEBRTC_LIBRARY_IMPL -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -I../../.. -Igen -I../../../third_party/abseil-cpp -I../../../third_party/libyuv/include -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -march=armv7-a -mfloat-abi=hard -mtune=generic-armv7-a -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -mfpu=neon -mthumb -Wall -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -Wno-comments -Wno-packed-not-aligned -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -std=gnu++14 -Wno-narrowing -Wno-class-memaccess -fno-exceptions -fno-rtti --sysroot=../../../build/linux/debian_sid_arm-sysroot -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../../../api/jsep_ice_candidate.cc -o obj/api/libjingle_peerconnection_api/jsep_ice_candidate.o
In file included from /mwon_disk/disk2/raspberry/build-webrtc/cross-pi-gcc-9.3.0-2/arm-linux-gnueabihf/include/c++/9.3.0/arm-linux-gnueabihf/bits/os_defines.h:39,
                 from /mwon_disk/disk2/raspberry/build-webrtc/cross-pi-gcc-9.3.0-2/arm-linux-gnueabihf/include/c++/9.3.0/arm-linux-gnueabihf/bits/c++config.h:524,
                 from /mwon_disk/disk2/raspberry/build-webrtc/cross-pi-gcc-9.3.0-2/arm-linux-gnueabihf/include/c++/9.3.0/bits/stl_algobase.h:59,
                 from /mwon_disk/disk2/raspberry/build-webrtc/cross-pi-gcc-9.3.0-2/arm-linux-gnueabihf/include/c++/9.3.0/memory:62,
                 from ../../../api/jsep_ice_candidate.h:19,
                 from ../../../api/jsep_ice_candidate.cc:11:
../../../build/linux/debian_sid_arm-sysroot/usr/include/features.h:446:12: fatal error: sys/cdefs.h: No such file or directory
  446 | #  include <sys/cdefs.h>
      |            ^~~~~~~~~~~~~
compilation terminated.
[2/2998] CXX obj/api/ice_transport_factory/ice_transport_factory.o
FAILED: obj/api/ice_transport_factory/ice_transport_factory.o 
arm-linux-gnueabihf-g++ -MMD -MF obj/api/ice_transport_factory/ice_transport_factory.o.d -DUSE_UDEV -DUSE_AURA=1 -DUSE_GLIB=1 -DUSE_NSS_CERTS=1 -DUSE_OZONE=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D_FORTIFY_SOURCE=2 -DCR_SYSROOT_HASH=3fcc1d4e44127006318371002a0f421a4fde2ab4 -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DWEBRTC_ENABLE_PROTOBUF=0 -DWEBRTC_INCLUDE_INTERNAL_AUDIO_DEVICE -DRTC_ENABLE_VP9 -DHAVE_SCTP -DWEBRTC_USE_H264 -DWEBRTC_ARCH_ARM -DWEBRTC_ARCH_ARM_V7 -DWEBRTC_HAS_NEON -DWEBRTC_LIBRARY_IMPL -DWEBRTC_NON_STATIC_TRACE_EVENT_HANDLERS=0 -DWEBRTC_POSIX -DWEBRTC_LINUX -DABSL_ALLOCATOR_NOTHROW=1 -I../../.. -Igen -I../../../third_party/abseil-cpp -I../../../third_party/libyuv/include -fno-strict-aliasing --param=ssp-buffer-size=4 -fstack-protector -funwind-tables -fPIC -pipe -pthread -march=armv7-a -mfloat-abi=hard -mtune=generic-armv7-a -Wno-builtin-macro-redefined -D__DATE__= -D__TIME__= -D__TIMESTAMP__= -mfpu=neon -mthumb -Wall -Wno-psabi -Wno-unused-local-typedefs -Wno-maybe-uninitialized -Wno-deprecated-declarations -Wno-comments -Wno-packed-not-aligned -Wno-missing-field-initializers -Wno-unused-parameter -O2 -fno-ident -fdata-sections -ffunction-sections -fno-omit-frame-pointer -g0 -fvisibility=hidden -std=gnu++14 -Wno-narrowing -Wno-class-memaccess -fno-exceptions -fno-rtti --sysroot=../../../build/linux/debian_sid_arm-sysroot -fvisibility-inlines-hidden -Wnon-virtual-dtor -Woverloaded-virtual -c ../../../api/ice_transport_factory.cc -o obj/api/ice_transport_factory/ice_transport_factory.o
In file included from /mwon_disk/disk2/raspberry/build-webrtc/cross-pi-gcc-9.3.0-2/arm-linux-gnueabihf/include/c++/9.3.0/arm-linux-gnueabihf/bits/os_defines.h:39,
                 from /mwon_disk/disk2/raspberry/build-webrtc/cross-pi-gcc-9.3.0-2/arm-linux-gnueabihf/include/c++/9.3.0/arm-linux-gnueabihf/bits/c++config.h:524,
                 from /mwon_disk/disk2/raspberry/build-webrtc/cross-pi-gcc-9.3.0-2/arm-linux-gnueabihf/include/c++/9.3.0/string:38,
                 from ../../../api/ice_transport_interface.h:14,
                 from ../../../api/ice_transport_factory.h:14,
                 from ../../../api/ice_transport_factory.cc:11:
[... cut ...]
                 from ../../../third_party/abseil-cpp/absl/types/optional.h:38,
                 from ../../../call/adaptation/encoder_settings.h:14,
                 from ../../../call/adaptation/encoder_settings.cc:11:
../../../build/linux/debian_sid_arm-sysroot/usr/include/limits.h:26:10: fatal error: bits/libc-header-start.h: No such file or directory
   26 | #include <bits/libc-header-start.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
```

mpromonet, 2020-05-03:
Hi miguelwon,

I am not sure it is building on Ubuntu 16.04 with this compiler.
The automatic build on this project use Ubuntu 18.04.

Best Regards,
Michel.

closed 2020-05-25 by mpromonet