mpromonet/webrtc-streamer #320: iOS resolution issue

SnoyIatK, 2020-05-01:
On iOS, play a streaming video in portrait mode always results in full screen height, and can not resize to fit screen resolution.
When playing same mp4 file, the player will only fit to screen size.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://webrtc-streamer.herokuapp.com/?layout=2x2' using any iPhone
2. Tap to play any video in portrait mode
3. Change to landscape mode

**Expected behavior**
The player should fit to device screen in portrait mode.
See below screenshots.

**Screenshots**
[image]
[image]
[image]
.

**Smartphone (please complete the following information):**
 - OS: iOS 13.3.1
 - Browser: Safari

mpromonet, 2020-07-04:
Hi SnoyIatK,

This seems html/js/css stuff, there is just very simple stuff to display webrtc streams.
You can easily do your own presnetaion using the webcomponent or the javascript interface or the http interface.

Best Regards,
Michel.

closed 2020-09-19 by mpromonet

jumarti, 2020-09-27:
This is a default behaviour on IOs devices. Set the HTML attribute `playsinline` to the video.  It worked for me