mpromonet/webrtc-streamer #322: Not working on iPhones

RobBiddle, 2020-05-13:
**Describe the bug**
Video does not play on iPhones. It does work on iPads. 

**Additional context**
This appears to be the problem:
https://forums.developer.apple.com/thread/88405

mpromonet, 2020-05-13:
Hi Robert,

There is no problem to add "playsinline" attributes to the video tag, this can be done in the HTML.

Best Regards,
Michel.

RobBiddle, 2020-05-14:
Thank you Michel, I'm using the Docker image and it is still not playing video when viewed from an iPhone.  Does the Docker image need to be updated as well?

FYI - There's a web-based iPhone emulator here: https://appetize.io/  The demo on the main page allows you to open Safari and browse to a WebRTC site to test.  I've also confirmed on a real iPhone.

closed 2021-07-04 by mpromonet