mpromonet/webrtc-streamer #324: Pointing a client to a remote stream

distr0, 2020-05-22:
I've been testing webrtc-streamer for a while on a local network, but I'm completely lost on how to point a client to the streamer when using a STUN/TURN server. 

I believe I would have to run it like this on the streamer side (where the camera is located):
```
 ./mpromonet/webrtc-streamer -s<user>:<pass>@<stun_server>:3478 -t<user>:<pass>@<tun_server>:3478 -u rtsp://user:pass@192.168.1.108:554/cam/realmonitor?channel=1&subtype=1
```
But I have no idea on how to point an instance of webrtc-streamer ( acting as a client and running elsewhere) to that stream.  
(is it possible?)

Could you provide a command line example on how to achieve this?

Thanks for your time

mpromonet, 2020-05-22:
Hi distr0,

The server side does not really needs to know what RTSP url to stream, it just make a dictionary available from http interface. You can ask to connect to an RTSP stream that is not defined in the backend.
For instance with an html `mycamera.html` like :
```
<html>
<head>
    <script type="module" src="webrtc-streamer-element.js"></script>
</head>
<body>
   <webrtc-streamer url="rtsp://user:pass@192.168.1.108:554/cam/realmonitor?channel=1&subtype=1"></webrtc-streamer>
</body>
</html>
```
Then connecting to http://127.0.0.1:8000/mycamera.html should display your camera, without giving it as a command line argument.

Best Regards,
Michel.

distr0, 2020-05-22:
Hi Michel,
> For instance with an html mycamera.html like 

Actually, I've already tried embedding it on a web page as is suggested on the readme, and it works just fine on a local network.

>  The STUN/TURN are retrieved from the client side using the api /api/getIceServers. This is implemented in WebRtcStreamer

This is what I'm lost at. How do I point an instance of webrtc-streamer to another using this api? Do I have to pass the external IP of the remote streamer to it somehow?

` ./mpromonet/webrtc-streamer -s<user>:<pass>@<stun_server>:3478 -t<user>:<pass>@<tun_server>:3478 -u rtsp://<remote_webrtc-streamer_external_IP>:<port>`

Something like that?

Many Thanks

mpromonet, 2020-05-22:
Hi,

The inputs streams of webrtc-streamer could be capture device, RTSP, screen, display, but not a WebRTC stream. Even this is probably possible, I don't figure out why doing this. 

If you are trying to access to webrtc-streamer running on LAN from a WAN, you need to configure NAT redirection  for tcp port for http (8000 by default) and it should be enouph as it seems you are using extenal STUN/TURN.

Best Regards,
Michel.

distr0, 2020-05-22:
Hi Michel,
I understand. My intention was to have a setup akin to a webrtc p2p video conference application, where the two parties (in this case a streamer and client) meet at a server and from then on they interchange data without opening ports, but then again I 'm new to webrtc so maybe I'm wrong about what's going on under the hood, and those two parties actually open ports to get through the NAT.

Anyways, now that I understand the whole scheme I'll have to try it and see if my mobile internet provider is preventing me from opening ports and reaching the streamer.

Thanks for your support!

closed 2020-05-22 by distr0