mpromonet/webrtc-streamer #328: audio streaming problem

VerHext, 2020-06-06:
First of all thanks for this nice project.

I try to play rtsp streams with webrtcstreamer.js in the browser. 
Video works fine. But it does not transmit audio.
I try several rtsp streamers, different codecs, demo video bunny, but nothing works for me.
That my client side: https://config.fail/uxewiqewec.html
Is there a special codec that must be used?

mpromonet, 2020-06-07:
Hi Johannes,

The audio support needs to have build with ALSA installed installed. During the cmake you should see `ALSA_FOUND=true`. There is no easy way to get this from the executable.
Something like `nm webrtc-streamer | grep -i alsa`

Next the format should be supported by live555 and WebRTC SDK, I made test with opus & pcm.
If the codec is not supported by WebRTC you should see the error `LiveAudioSource::onNewSession not support codec`    

Best Regards,
Michel.

VerHext, 2020-06-07:
Hello Michel,

thanks for your fast response. 

ALSA is installed and enabled. 
With Opus 48k 2 - I get a "not supported codec" error. Using PCM it doesn't play either.
https://config.fail/ovopupulec.md

mpromonet, 2020-06-07:
Hi Johannes,

There is the Describe answer for video but it seems the RTSP negociation for audio fails `Failed to DESCRIBE: Received a complete`. 
Still there is no log `LiveAudioSource::onNewSession` and then `LiveAudioSource::onData`, no data is received by the consumer.

Best Regards,
MicheL

VerHext, 2020-06-07:
oh yeah, sorry. 
Thanks works great!

closed 2020-06-07 by VerHext