mpromonet/webrtc-streamer #330: Can't hear audio when streaming file to browser

tcjohnson123, 2020-06-18:
**Description**

Really impressed with this project - great work!
[out.txt](https://github.com/mpromonet/webrtc-streamer/files/4799828/out.txt)

However, I can't hear audio when I stream to the browser.

I can't see what's wrong in the log file:

https://paste.ee/p/cJy8p

**To Reproduce**

Steps to reproduce the behavior:
1. Start the server : `webrtc-streamer -w html`
2. Open url : `http://127.0.0.1:8000/webrtcstreamer.html?video=file:///tcj/test.mkv&audio=file:///tcj/test.mkv`
3. I can't hear any audio.

If I play the file in Windows 'Films and TV' I can hear the audio. 

Note that the file is an mkv which I created with ffmpeg using h264 baseline and opus audio.

**Environment**
 - OS: Windows x64
 - Browser : Chrome and Edge tested (same problem)
 - Version:v0.2.6-32-g97644b1/Windows-AMD64 civetweb@v1.11 webrtc@de2049e85a-dirty live555helper@b4d86f2

**Additional context**

I've looked in the log file:

https://paste.ee/p/cJy8p

It is decoding audio packets but I can't hear anything.

I also get the same problem on Linux (Ubuntu 18.04).

tcjohnson123, 2020-06-26:
If I look at chrome://webrtc-internals/, it's receiving an audio stream in opus format:

[image]

I guess either the stream it's receiving is silence or the browser is applying a negative gain.

mpromonet, 2020-12-17:
Hi tcjohnson123, 

Previously the sound was muted without way to unmute it using the embeded html/js.
Since https://github.com/mpromonet/webrtc-streamer/commit/8df2f353474a59bbd3e84ee80f8c186b5681bc34, it should be possible to unmute. 
Starting mute allow to autostart stream.
Let me know if you still have problem with latest version.

Best Regards,
Michel.

tcjohnson123, 2020-12-17:
Hi Michel,

Thanks for the update.  

I have just tried webrtc-streamer-v0.3.3-17-g5e5933f-Windows-AMD64-Release Appveyor build x64 release. 

I followed my own steps listed above.  

I was able to see mute / unmute controls in the video control and change the volume slider.  However, I didn't hear any audio. When I looked in chrome://webrtc-internals I noticed that there were no audio packets reported in the RTCInboundRTPAudioStream statistics.  This is different to the behaviour I saw in June (you can see the screenshot above with the audio packets arriving successfully).  

So I'm afraid I'm unable to confirm your fix because I can't recreate the conditions from six months ago when I was seeing audio packets arriving.  I had to make the test.mkv file again with ffmpeg (using h264 / opus as I described in the original bug report) as I don't still have the file I used in June.  Something may have changed since such as a Windows 10 update or a change in ffmpeg.  

If I can find a test file that works, I'll let you know.  I can send you my test file if you want.  

It's not causing me a problem at the moment so feel free to close the issue if you don't think it is important. 

Thanks,

Tom

closed 2021-07-04 by mpromonet