mpromonet/webrtc-streamer #332: how to download the laster tag Windows-AMD64-Release.tar.gz

dolwin, 2020-07-16:
I am not good at C language development, I downloaded the old version and have problems

mpromonet, 2020-07-16:
Hi Dolwin,

What kind of problem do you have ?

Best Regards,
Michel.

dolwin, 2020-07-17:
@mpromonet  thanks you for reply
can you send me the laster Windows-AMD64-Release.tar.gz to test，I tried one night and it didn't build successfully
my email is glasssky001@gmail.com
sorry to bother you

mpromonet, 2020-07-18:
Hi,
I published a new release, you can get https://github.com/mpromonet/webrtc-streamer/releases/download/v0.3.0/webrtc-streamer-v0.3.0-Windows-AMD64-Release.tar.gz
Last build can be get from appveyor https://ci.appveyor.com/project/mpromonet/webrtc-streamer/builds/34171553/job/syveb2lvr9l6csgg/artifacts

Best Regards,
Michel

dolwin, 2020-07-20:
@mpromonet  thanks you

closed 2020-07-26 by mpromonet