mpromonet/webrtc-streamer #333: Can't build last version

ArtemM, 2020-07-16:
**Describe the bug**
I've tried to install the last version from the repository but getting the error on step "cmake . && make". Possibly something in dependencies had been changed?

```console
[ 95%] Building CXX object CMakeFiles/webrtc-streamer.dir/src/PeerConnectionManager.cpp.o
/home/admin/webrtc-streamer/src/PeerConnectionManager.cpp: In member function 'bool PeerConnectionManager::AddStreams(webrtc::PeerConnectionInterface*, const string&, const string&, const string&)':
/home/admin/webrtc-streamer/src/PeerConnectionManager.cpp:1036:3: error: 'BitrateParameters' is not a member of 'webrtc::PeerConnectionInterface'
   webrtc::PeerConnectionInterface::BitrateParameters bitrateParam;
   ^~~~~~
/home/admin/webrtc-streamer/src/PeerConnectionManager.cpp:1037:3: error: 'bitrateParam' was not declared in this scope
   bitrateParam.min_bitrate_bps = absl::optional<int>(bitrate / 2);
   ^~~~~~~~~~~~
CMakeFiles/webrtc-streamer.dir/build.make:94: recipe for target 'CMakeFiles/webrtc-streamer.dir/src/PeerConnectionManager.cpp.o' failed
make[2]: *** [CMakeFiles/webrtc-streamer.dir/src/PeerConnectionManager.cpp.o] Error 1
CMakeFiles/Makefile2:118: recipe for target 'CMakeFiles/webrtc-streamer.dir/all' failed
make[1]: *** [CMakeFiles/webrtc-streamer.dir/all] Error 2
```

Full log https://gist.github.com/ArtemM/66af383b869636596e3ba4011c68df2c

**To Reproduce**
Steps to reproduce the behavior:
Follow https://github.com/mpromonet/webrtc-streamer/blob/master/README.md "Build" section

**Expected behavior**
Application built without errors

**Desktop (please complete the following information):**
 - OS:
Two tries: on Amazon Linux 2 and Debian

 - Version 
checkout from repository

mpromonet, 2020-07-16:
Hi ArtemM,

Probably `webrtc::PeerConnectionInterface::BitrateParameters` changed or disappear, you can remove the related code. I will look to this.

Best Regards,
Michel.

ArtemM, 2020-07-16:
Thanks Michel, trying now (will try quick-hack replacing with BitrateSettings). Meanwhile yes, looks like webrtc library had been adjusted as per 
https://code.immersaview.com/remotes/webrtc/commit/0c4f7beb25ad062e17b1cd26d5e0e3c40ab00f5c

ArtemM, 2020-07-16:
Ah, I see you did the "quick hack" too :) . Trying a new version, thanks for quick check!

ArtemM, 2020-07-16:
>Built target webrtc-streamer

closed 2020-07-16 by ArtemM