mpromonet/webrtc-streamer #334: Default html and demos are not working in Chrome

ArtemM, 2020-07-16:
**Describe the bug**
In chrome video does not load in demo (https://webrtc-streamer.herokuapp.com/) nor in simple test run: (e.g. ./webrtc-streamer rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov). 
FF is fine.

**To Reproduce**
Steps to reproduce the behavior:
go to https://webrtc-streamer.herokuapp.com/ , check gray screen

Weird thing is that while code seems to be the same for online demo and hosted HTML page on online demo I don't get any JS errors in console (just gray video screen) while running hosted version gives:
```console
Uncaught (in promise) TypeError: Failed to execute 'addIceCandidate' on 'RTCPeerConnection': Candidate missing values for both sdpMid and sdpMLineIndex
    at RTCPeerConnection.withCallback [as addIceCandidate] (adapter.min.js:5)
    at WebRtcStreamer.onReceiveCandidate (webrtcstreamer.js:289)
    at webrtcstreamer.js:138
    at tryCallOne (request.min.js:10)
    at request.min.js:10
    at MutationObserver.flush (request.min.js:27)
```

**Expected behavior**
video played

**Desktop (please complete the following information):**
 - OS: 
macOS Catalina
 - Browser
Chrome 
 - Version 
Version 84.0.4147.89 (Official Build) (64-bit)

**Additional context**
Potentially relevant issue: https://bugs.chromium.org/p/chromium/issues/detail?id=978582

mpromonet, 2020-07-17:
Hi ArtemM,

I just tried with the same version of Chrome on windows, videos are displayed.
The error you are pointing is (probably) the end of candidate event that use an empty candidate.
This may be related to MacOS or the STUN availibility, looking to the ICE candidate may help you to understand. If there is no ICE candidate valid (with your public IP) the video cannot be displayed.

Best Regards,
Michel.

ArtemM, 2020-07-17:
Hello Michel!

> related to MacOS 

If it is OS-specific then probably FF would not work too but it works. 

Possibly rather related to Chrome version? I've checked on wife's laptop under Windows and Chrome version 83.0.4103.116 with the same result 

[image]

Also addIceCandidate by itself works, it is syntax issue as there is no standard handling of end of candidates so specifically addIceCandidate() applied without parameters throws error. I've tried several other variations like addIceCandidate(null), addIceCandidate({}), addIceCandidate(null, null), addIceCandidate('') but nothing worked for me.

Regards,
Artem.

mpromonet, 2020-07-17:
Hi,

By the way `Prib` is not working, better to try with Bunny http://webrtc-streamer.herokuapp.com/webrtcstreamer.html?video=Bunny&options=rtptransport%3Dtcp%26timeout%3D60

Best Regards,
Michel.

ArtemM, 2020-07-17:
Michel, 

Bunny behaves in the same way on two machines tested. I moved further and checked behavior in BrowserStack. Error by itself is present in any chrome I checked (75+) **but** versions prior 77 (e.g. 76, 75, etc) are still able to play video. Version 78-85 are not able to play video (which likely eliminates OS and STUN relation of the issue as browserstack environment should be the same). I've also tried under VPN just to adjust something in network environment. 
The weird part: two friends of mine win7 chrome 84 and some win chrome 83 were able to open bunny without issues. 

Assuming error is present in BrowserStack to running older versions of Chrome (and video plays) there is a chance that gray screen instead of video is not directly related to the error in console (empty addIceCandidate) and/or somehow related to the execution order.

If no other reports observed I would suggest to check in browserstack on your side or just ignore for now. I'll try to check from a different network, maybe something messes up things in mine. I also tried and observed same behavior with VPN just to adjust network config.

Regards,
Artem.

lukelu0520, 2020-08-13:
I got the similar issue ( #346 ). Instead of streaming a third-party source, you could try stream your own source such as screen://...  or window://.. to see if the issue is still there.

closed 2020-12-17 by mpromonet