mpromonet/webrtc-streamer #335: webrtc-streamer in docker with audio

rez0n, 2020-07-17:
Hello, advice please how to restream with audio support using docker image.

docker parameters
```
docker run --name webrtc-streamer --rm --net=host -it mpromonet/webrtc-streamer rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov
```
and gets
```
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL
[000:003] [1] (audio_mixer_manager_alsa_linux.cc:173): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:004] [1] (audio_device_alsa_linux.cc:1353): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM
[000:004] [1] (audio_device_alsa_linux.cc:801): unable to open playback device: No such file or directory (-2)
[000:004] [1] (audio_device_alsa_linux.cc:1353): GetDevicesInfo - Could not find device name or numbers
```

I'll try to pass `--device /dev/snd` to the docker, but it does not help. Do I need to have a physical soundcard to restream?

mpromonet, 2020-07-17:
Hi,

This is not related of audio capture, probably the RTSP audio stream is not supported.
Only a few are supported PCM & OPUS

Best Regards,

Michel.

rez0n, 2020-07-17:
Hi, Michel! 
Thanks for your answer.
On demo https://webrtc-streamer.herokuapp.com/Bunny.html is no audio, it is ok because the audio codec is unsupported?

mpromonet, 2020-07-17:
Hi Denis,

Audio of rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov is AAC, there is probably an error in the log indicating audio format is not supported.

Best Regards,
Michel.

closed 2020-07-17 by rez0n