mpromonet/webrtc-streamer #338: CMake Error (WEBRTC_LIBRARY) on Ubuntu 18.04 aarch64 (arm64)

lukelu0520, 2020-07-27:
Dear webrtc-streamer friends,

**Describe the bug**
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
WEBRTC_LIBRARY
    linked by target "webrtc-streamer" in directory /webrtc-streamer

**To Reproduce**
Error happens when doing 
cmake -DCMAKE_SYSTEM_PROCESSOR=${ARCH} -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_C_COMPILER=aarch64-linux-gnu-gcc -DCMAKE_CXX_COMPILER=aarch64-linux-gnu-g++ -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=NEVER -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_PACKAGE=ONLY -DWEBRTCDESKTOPCAPTURE=OFF . && make

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04-Mate running on Raspberry Pi 4 8GB

**Additional context**
I simply follow the RUN part of Dockerfile.arm64.
Everything goes well on ubuntu 18.04 amd64 (by simply follow RUN part of Dockerfile), but fail in aarch64 (arm64). Hope to hear your expertise.

mpromonet, 2020-07-27:
Hi lukelu0520,

This may be if you run cmake interrupt it and rerun. Doing this the cmake process generated the file with WebRTC argument and use this file to know if WebRTC was built. In such a case remove `webrtc/src/out` folder and run cmake again.   
It is also possible WebRTC build fails but based on the information you gave, it is not possible to know.

Best Regards,
Michel.

lukelu0520, 2020-07-27:
Hi Michel,

Thanks for your prompt reply. After comparing results between amd64 and arm64, we found the reason is the WebRTC build fails. During WebRTC build for arm64, there is an issue on gn.py in depot_tools: 

Traceback (most recent call last):
  File "/depot_tools/gn.py", line 76, in <module>
    sys.exit(main(sys.argv))
  File "/depot_tools/gn.py", line 71, in main
    return subprocess.call([gn_path] + args[1:])
  File "/usr/lib/python2.7/subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error

=====

We believe this error causes the WebRTC build fail; were wondering if you have any idea on it.

Thank you
-- Luke

lukelu0520, 2020-07-28:
Unfortunately, cross-compile fail in the final step:  [100%] Linking CXX executable webrtc-streamer

/webrtc/src/out/arm64/obj/third_party/jsoncpp/jsoncpp/json_reader.o: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
CMakeFiles/webrtc-streamer.dir/build.make:303: recipe for target 'webrtc-streamer' failed
make[2]: *** [webrtc-streamer] Error 1
CMakeFiles/Makefile2:105: recipe for target 'CMakeFiles/webrtc-streamer.dir/all' failed
make[1]: *** [CMakeFiles/webrtc-streamer.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

============================

Was wondering if anyone gets the same issue with us.

Best,
-- Luke

lukelu0520, 2020-07-28:
Problem solved by changing line 22 of Dockerfile.arm64:  ${ARCH} to "arm64"

Eager to put it on my aarch64+Ubuntu 18.04+Raspberry Pi 4 to see experience webrtc-streamer !

closed 2020-07-28 by lukelu0520

mpromonet, 2020-07-28:
Hi lukelu0520,

I am glad you found the problem.
As I understand, you did not set ARCH variable running cmake with  `-DCMAKE_SYSTEM_PROCESSOR=${ARCH}`. In the docker this is set by the arg `ARCH` that is by default set to arm64.
I make it run on a Raspberry2, probably it give more power with a Raspberry4...
At the origin the aarch64 build was done to give a exemple of building for arm64, I don't know if it is a good cross-compiler for Raspberry 4 ?

Best Regards,
Michel.

lukelu0520, 2020-07-28:
Hi Michel, 

Thanks for your dedicated efforts on webrtc-streamer. Yes, raspberry pi makes a big leap on 4th generation (aka RP4). It is now able to run 64bit OS (e.g. Ubuntu 18.04) with its arm64 (v8) cpu using 8GB RAM. We are running ubuntu 18.04-mate smoothly; to me, it's working as a light-weight desktop. And the community keeps improving both firmware and software. 

Also, the only reason we choose Ubuntu rather than Raspbian is because of Robot Operating System (ROS). The ROS is most supportive on Ubuntu for a robotic prototyping system.

After the cross-compilation, we copied app folder to RP4, and run ./webrtc-streamer -V on RP4, it prints: 
v0.3.0-dirty/Linux-arm64 civetweb@v1.11 webrtc@0bb2639060-dirty live555helper@b4d86f2

and then tried ./webrtc-streamer rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov 
it outputs:

Version:v0.3.0-dirty/Linux-arm64 civetweb@v1.11 webrtc@0bb2639060-dirty live555helper@b4d86f2
{
	"urls" : 
	{
		"rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov" : 
		{
			"video" : "rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov"
		}
	}
}Logger level:3
HTTP Listen at 0.0.0.0:8000

Good news is that we open a browser on a machine rather than RP4, and try: http://_${RP's LAN IP}_:8000/, we see the BigBuckBunny_115k video.

Do you have a suggestion on further testing to validate the performance toward low-latency live streaming using RP camera ? 
We hope to do more works and keep you posted !  

-- Luke