mpromonet/webrtc-streamer #339: How to setup mic audio with a V4L2 device?

ErikAndren, 2020-07-28:
**Describe the bug**
I'm trying to stream a birdnest with a webcam with an embedded microphone using raspian stretch.
The device is uvc-compatible and I am able to see the video stream using webrtc-streamer.
However I am not picking up any audio and I don't understand how to configure webrtc-streamer to support it.
I know that the mic is working as I am able to record audio using alsa and stream audio separately using ffmpeg.

The instructions says you to supply the -a switch but I am unsure what to do next.
No audio is arriving in the browser.

I am currently using the v0.2.6 build as I was unable to checkout webrtc as it failed and complained about missing armv6l components.

**To Reproduce**
Steps to reproduce the behavior:
/webrtc-streamer -vvv -a /dev/video0 

**Expected behavior**
Getting video and audio stream in browser

**Desktop (please complete the following information):**
 - OS: Raspian Buster, 
 - Hardware: Raspberry pi version 1

**Additional context**

Log:
Version:v0.2.6/Linux-armv6l civetweb@v1.11 webrtc@d9ebe01540-dirty live555helper@b4d86f2
{
	"urls" : 
	{
		"/dev/video0" : 
		{
			"video" : "/dev/video0"
		}
	}
}Logger level:0
[000:000] [1408] (audio_processing_impl.cc:320): Injected APM submodules:
Echo control factory: 0
Echo detector: 0
Capture analyzer: 0
Capture post processor: 0
Render pre processor: 0
[000:008] [1408] (webrtc_voice_engine.cc:206): WebRtcVoiceEngine::WebRtcVoiceEngine
[000:010] [1408] (webrtc_video_engine.cc:484): WebRtcVideoEngine::WebRtcVideoEngine()
[000:016] [1408] (webrtc_voice_engine.cc:229): WebRtcVoiceEngine::Init
[000:019] [1408] (webrtc_voice_engine.cc:237): Supported send codecs in order of preference:
[000:024] [1408] (webrtc_voice_engine.cc:240): opus/48000/2 { minptime=10 useinbandfec=1 } (111)
[000:026] [1408] (webrtc_voice_engine.cc:240): ISAC/16000/1 (103)
[000:029] [1408] (webrtc_voice_engine.cc:240): G722/8000/1 (9)
[000:030] [1408] (webrtc_voice_engine.cc:240): ILBC/8000/1 (102)
[000:031] [1408] (webrtc_voice_engine.cc:240): PCMU/8000/1 (0)
[000:032] [1408] (webrtc_voice_engine.cc:240): PCMA/8000/1 (8)
[000:033] [1408] (webrtc_voice_engine.cc:240): CN/16000/1 (105)
[000:033] [1408] (webrtc_voice_engine.cc:240): CN/8000/1 (13)
[000:035] [1408] (webrtc_voice_engine.cc:240): telephone-event/48000/1 (110)
[000:035] [1408] (webrtc_voice_engine.cc:240): telephone-event/16000/1 (113)
[000:036] [1408] (webrtc_voice_engine.cc:240): telephone-event/8000/1 (126)
[000:036] [1408] (webrtc_voice_engine.cc:243): Supported recv codecs in order of preference:
[000:038] [1408] (webrtc_voice_engine.cc:246): opus/48000/2 { minptime=10 useinbandfec=1 } (111)
[000:039] [1408] (webrtc_voice_engine.cc:246): ISAC/16000/1 (103)
[000:039] [1408] (webrtc_voice_engine.cc:246): G722/8000/1 (9)
[000:040] [1408] (webrtc_voice_engine.cc:246): ILBC/8000/1 (102)
[000:040] [1408] (webrtc_voice_engine.cc:246): PCMU/8000/1 (0)
[000:040] [1408] (webrtc_voice_engine.cc:246): PCMA/8000/1 (8)
[000:041] [1408] (webrtc_voice_engine.cc:246): CN/16000/1 (105)
[000:042] [1408] (webrtc_voice_engine.cc:246): CN/8000/1 (13)
[000:042] [1408] (webrtc_voice_engine.cc:246): telephone-event/48000/1 (110)
[000:043] [1408] (webrtc_voice_engine.cc:246): telephone-event/16000/1 (113)
[000:044] [1408] (webrtc_voice_engine.cc:246): telephone-event/8000/1 (126)
[000:045] [1408] (webrtc_voice_engine.cc:316): WebRtcVoiceEngine::ApplyOptions: AudioOptions {aec: 1, agc: 1, ns: 1, hf: 1, swap: 0, audio_jitter_buffer_max_packets: 200, audio_jitter_buffer_fast_accelerate: 0, audio_jitter_buffer_min_delay_ms: 0, audio_jitter_buffer_enable_rtx_handling: 0, typing: 1, experimental_agc: 0, experimental_ns: 0, residual_echo_detector: 1, }
[000:047] [1408] (webrtc_voice_engine.cc:432): Stereo swapping enabled? 0
[000:047] [1408] (webrtc_voice_engine.cc:437): NetEq capacity is 200
[000:048] [1408] (webrtc_voice_engine.cc:443): NetEq fast mode? 0
[000:049] [1408] (webrtc_voice_engine.cc:449): NetEq minimum delay is 0
[000:049] [1408] (webrtc_voice_engine.cc:455): NetEq handle reordered packets? 0
[000:050] [1408] (webrtc_voice_engine.cc:467): Experimental ns is enabled? 0
[000:050] [1408] (webrtc_voice_engine.cc:518): NS set to 1
[000:050] [1408] (webrtc_voice_engine.cc:522): Typing detection is enabled? 1
[000:052] [1408] (audio_processing_impl.cc:574): AudioProcessing::ApplyConfig: AudioProcessing::Config{ pipeline: {maximum_internal_processing_rate: 32000, multi_channel_render: 0, , multi_channel_capture: 0}, pre_amplifier: { enabled: 0, fixed_gain_factor: 1 }, high_pass_filter: { enabled: 1 }, echo_canceller: { enabled: 1, mobile_mode: 0, enforce_high_pass_filtering: 1 }, noise_suppression: { enabled: 1, level: High }, transient_suppression: { enabled: 0 }, voice_detection: { enabled: 1 }, gain_controller1: { enabled: 1, mode: AdaptiveAnalog, target_level_dbfs: 3, compression_gain_db: 9, enable_limiter: 1, analog_level_minimum: 0, analog_level_maximum: 255 }, gain_controller2: { enabled: 0, fixed_digital: { gain_db: 0 }, adaptive_digital: { enabled: 0, level_estimator: Rms, use_saturation_protector: 1, extra_saturation_margin_db: 2 } }, residual_echo_detector: { enabled: 1 }, level_estimation: { enabled: 0 } }
[000:056] [1408] (render_delay_buffer.cc:362): Applying total delay of 5 blocks.
[000:059] [1408] (matched_filter.cc:450): Filter 0: start: 0 ms, end: 128 ms.
[000:061] [1408] (matched_filter.cc:450): Filter 1: start: 96 ms, end: 224 ms.
[000:062] [1408] (matched_filter.cc:450): Filter 2: start: 192 ms, end: 320 ms.
[000:063] [1408] (matched_filter.cc:450): Filter 3: start: 288 ms, end: 416 ms.
[000:063] [1408] (matched_filter.cc:450): Filter 4: start: 384 ms, end: 512 ms.
[000:066] [1408] (agc_manager_direct.cc:69): [agc] GetMinMicLevel
[000:068] [1408] (agc_manager_direct.cc:73): [agc] Using default min mic level: 12
HTTP Listen at 0.0.0.0:8000

mpromonet, 2020-07-28:
Hi Erik,

Probably the problem you are speaking about component, is about glic or libstdc++ release ?
I have some difficulty to have static link solution building for arm & x86.

In order to know if the sound input is detected, you could call the api `/api/getAudioDeviceList`. It should give the name of the audio device that you can then use in the audio url.

Best Regards,

Michel.

ErikAndren, 2020-07-28:
Hi and thank you for your help.
I have multipe issues:
When trying to fetch webrtc I get the following error:

erik@Phobos:~/Development/webrtc $ fetch --no-history webrtc
Errors:
  failed to resolve infra/3pp/tools/git/linux-armv6l@version:2.24.1.chromium.5 (line 27): no such package: infra/3pp/tools/git/linux-armv6l
  failed to resolve infra/3pp/tools/cpython/linux-armv6l@version:2.7.17.chromium.22 (line 21): no such package: infra/3pp/tools/cpython/linux-armv6l
  failed to resolve infra/3pp/tools/cpython3/linux-armv6l@version:3.8.0.chromium.8 (line 24): no such package: infra/3pp/tools/cpython3/linux-armv6l
/home/erik/Development/depot_tools/bootstrap_python3: line 32: bootstrap-3.8.0.chromium.8_bin/python3/bin/python3: No such file or directory
cat: /home/erik/Development/depot_tools/python3_bin_reldir.txt: No such file or directory
[E2020-07-28T22:11:47.499320+02:00 1745 0 annotate.go:241] original error: fork/exec /home/erik/Development/depot_tools/python3: no such file or directory
[E2020-07-28T22:11:47.508508+02:00 1745 0 annotate.go:241] 
[E2020-07-28T22:11:47.510763+02:00 1745 0 annotate.go:241] goroutine 1:
[E2020-07-28T22:11:47.513025+02:00 1745 0 annotate.go:241] #0 go.chromium.org/luci/vpython/python/interpreter.go:89 - python.(*Interpreter).GetVersion()
[E2020-07-28T22:11:47.514523+02:00 1745 0 annotate.go:241] #1 go.chromium.org/luci/vpython/venv/config.go:286 - venv.(*Config).resolvePythonInterpreter()
[E2020-07-28T22:11:47.516389+02:00 1745 0 annotate.go:241]   reason: failed to determine Python version for: /home/erik/Development/depot_tools//python3
[E2020-07-28T22:11:47.517854+02:00 1745 0 annotate.go:241] 
[E2020-07-28T22:11:47.519345+02:00 1745 0 annotate.go:241] #2 go.chromium.org/luci/vpython/venv/config.go:186 - venv.(*Config).makeEnv()
[E2020-07-28T22:11:47.519996+02:00 1745 0 annotate.go:241]   reason: failed to resolve system Python interpreter
[E2020-07-28T22:11:47.521538+02:00 1745 0 annotate.go:241] 
[E2020-07-28T22:11:47.522587+02:00 1745 0 annotate.go:241] #3 go.chromium.org/luci/vpython/venv/venv.go:150 - venv.With()
[E2020-07-28T22:11:47.523140+02:00 1745 0 annotate.go:241]   reason: failed to initialize empty probe environment
[E2020-07-28T22:11:47.523787+02:00 1745 0 annotate.go:241] 
[E2020-07-28T22:11:47.524305+02:00 1745 0 annotate.go:241] #4 go.chromium.org/luci/vpython/run.go:62 - vpython.Run()
[E2020-07-28T22:11:47.524820+02:00 1745 0 annotate.go:241] #5 go.chromium.org/luci/vpython/application/application.go:320 - application.(*application).mainImpl()
[E2020-07-28T22:11:47.526351+02:00 1745 0 annotate.go:241] #6 go.chromium.org/luci/vpython/application/application.go:406 - application.(*Config).Main.func1()
[E2020-07-28T22:11:47.528988+02:00 1745 0 annotate.go:241] #7 go.chromium.org/luci/vpython/application/support.go:46 - application.run()
[E2020-07-28T22:11:47.530674+02:00 1745 0 annotate.go:241] #8 go.chromium.org/luci/vpython/application/application.go:405 - application.(*Config).Main()
[E2020-07-28T22:11:47.532275+02:00 1745 0 annotate.go:241] #9 vpython/main.go:106 - main.mainImpl()
[E2020-07-28T22:11:47.533593+02:00 1745 0 annotate.go:241] #10 vpython/main.go:112 - main.main()
[E2020-07-28T22:11:47.534873+02:00 1745 0 annotate.go:241] #11 runtime/proc.go:203 - runtime.main()
[E2020-07-28T22:11:47.536320+02:00 1745 0 annotate.go:241] #12 runtime/asm_arm.s:868 - runtime.goexit()

When downloading and running the armv6l 0.3.0 release I get the following error:
erik@Phobos:~/Development/webrtc-streamer-v0.3.0-Linux-armv6l-Release $ ./webrtc-streamer 
./webrtc-streamer: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by ./webrtc-streamer)

This is due to that raspbian buster does not support >= GLIBCXX_3.4.26:

erik@Phobos:~/Development/webrtc-streamer-v0.3.0-Linux-armv6l-Release $ strings /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 | grep GLIBC
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
GLIBCXX_3.4.23
GLIBCXX_3.4.24
GLIBCXX_3.4.25
GLIBC_2.4
GLIBC_2.18
GLIBC_2.16
GLIBC_2.17
GLIBCXX_DEBUG_MESSAGE_LENGTH

ErikAndren, 2020-07-28:
getAudioDeviceList is just empty 

uri:/api/getAudioDeviceList
answer:[]

mpromonet, 2020-07-28:
If getAudioDeviceList is empty, it means the audio capture is not detected.
You may try different audio layer, but maybe it needs to build WebRTC, ALSA device should be supported

closed 2020-09-19 by mpromonet