mpromonet/webrtc-streamer #340: unable to start program

whucj, 2020-07-29:
On the debug model，the below error  occures.
[image]

My computer environment is as follows:
webrtc-streamer:master branch
webrtc version:m84

windows 10

visual studio 2017

windows 10 sdk  10.0.19041

Thanks

mpromonet, 2020-07-30:
Hi,

I know you are right since I moved the build from visual studio solution to ninja, it has never worked in debug on windows.

Best Regards,
Michel.

mpromonet, 2020-09-22:
Hi,

If you like, building with 
```
cmake -G "Visual Studio 16 2019 Win64" -DWEBRTCBUILD=DEBUG .
msbuild webrtc-streamer.sln -p:Configuration=Debug
```
This allow to build a runnable executable and have debug activities.

Best Regards,
Michel.

Teddy2212, 2020-09-28:
how to fix it CMake Error and Cmake generate step failed....
[image]

mpromonet, 2021-01-24:
Hi,

I am closing the issue as it is working now, using https://github.com/mpromonet/webrtc-streamer/releases/download/v0.3.5/webrtc-streamer-v0.3.5-Windows-AMD64-Debug.tar.gz.

Best Regards,
Michel.

closed 2021-01-24 by mpromonet