mpromonet/webrtc-streamer #341: External can not access webrtc stream even port forwarding is set

lukelu0520, 2020-07-29:
Dear webrtc-streamer friends,

I've set a port forwarding on port 8080 to the machine host webrtc stream. Computers in the same local area could access the stream great by http://_my_public_ip_:8080   
However, external visitor still can not get webrtc stream access using the same address.  That's weird, because external visitors are able to access my website by http://_my_public_ip_:80  after I set a port forwarding on port 80.

Maybe I miss something critical on webrtc principle, hope to hear your expertise.

-- Luke

mpromonet, 2020-07-30:
Hi Luke,

This depends on ICE negociation, and if you can reach the STUN server, by default it is set to `stun.l.google.com:19302`
This https://stackoverflow.com/questions/35434439/c-get-my-public-ip-address/35437308#35437308 explain how to test it.

If you cannot reach the STUN server you can use the embeded one, exposing it, some information are in https://github.com/mpromonet/webrtc-streamer#using-embedded-stunturn-server-behind-a-nat

Best Regards,
Michel.

lukelu0520, 2020-07-30:
Hi Michel,

Thanks for your reply. We are currently setting a DMZ for the RP4 with specific port forwarding to solve the problem.
STUN or TURN is a more secure solution and we would start adapting it in the near future.  

BTW, why you recommend STUN rather than TURN ? To my knowledge, STUN is limited in UDP.

-- Luke

closed 2020-07-30 by lukelu0520