mpromonet/webrtc-streamer #342: Fatal error in: ../../rtc_base/synchronization/mutex.h, line 72

RobBiddle, 2020-08-07:
**Describe the bug**
Using Docker image running on EC2 Amazon Linux host
I'm not sure when this broke exactly but it would have been fairly recently. I've been using the docker image for a long time now to stream web cams, but the container won't work anymore using the latest image.

The docker service log shows:
```
| }Logger level:3
| [000:000] [1] (audio_device_alsa_linux.cc:1353): GetDevicesInfo - Could not find device name or numbers
| ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL
| [000:001] [1] (audio_mixer_manager_alsa_linux.cc:173): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
| [000:001] [1] (adm_helpers.cc:48): Unable to access speaker.
|
|
| #
| # Fatal error in: ../../rtc_base/synchronization/mutex.h, line 72
| # last system error: 0
| # Check failed: holder != current (140231100124928 vs. 140231100124928)
```

RobBiddle, 2020-08-07:
Figured out that adding `-a` to the end of the docker command would disable audio.
Something changed to cause that to become necessary, I've been running this without `-a` for over a year without issue.

mpromonet, 2020-08-08:
Hi Robert,

I will look if I can repdroduce the problem, anyway running with `-a` option in a container seems a good option (there is probably no capture audio device)

Best Regards,
Michel

bkanuka, 2020-08-08:
I can confim that running the docer container with `-a` fixes this for me also.

mpromonet, 2020-08-08:
Hi,

I just give a try with docker images latest (v0.2.8-19-g7fbc2ed) and v0.3.0 and it runs without problem. Maybe this is depending on host OS version ? or docker options ?

Best Regards,
Michel.

mpromonet, 2021-01-09:
Hi,

The commit https://github.com/mpromonet/webrtc-streamer/commit/09deef002ce70f9c4fa90fe02297058a8b57db54 should fix this problem.

In case of new problem don't hesitate to open a new issue.

Best Regards,
Michel.

closed 2021-01-09 by mpromonet