mpromonet/webrtc-streamer #343: v0.3.0 Raspberry Pi  libstdc++.so.6: version `GLIBCXX_3.4.26' not found

obalyuk87, 2020-08-09:
**Describe the bug**
Published version of v0.3.0 does not run on Raspberry Pi. 

Crashes with error message 
```
 ./webrtc-streamer: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by ./webrtc-streamer)
```

Probably related to: https://github.com/mpromonet/webrtc-streamer/issues/258

**To Reproduce**
Steps to reproduce the behavior:
* Option 1: run docker version on  Raspberry Pi 
  `FROM mpromonet/webrtc-streamer:v0.3.0-armv7`... fails
   Note that `mpromonet/webrtc-streamer:v0.2.7-armv7` works fine

* Option 2: download armv7 static binary from releases and try to run binary.
[image]
```
 cd /tmp/
wget https://github.com/mpromonet/webrtc-streamer/releases/download/v0.3.0/webrtc-streamer-v0.3.0-Linux-armv7l-Release.tar.gz
tar xzfv webrtc-streamer-v0.3.0-Linux-armv7l-Release.tar.gz
cd webrtc-streamer-v0.3.0-Linux-armv7l-Release/
./webrtc-streamer
```

**Expected behavior**
should work without error :-)

**Desktop (please complete the following information):**
 - Raspbian GNU/Linux 10 (buster)
 
**Additional context**
AMAZING PROJECT! Thank you!

mpromonet, 2020-08-09:
Hi obalyuk87 ,

You are right I recently remove the fix introduce for #258 due to conflict between CXX lib use by WebRTC SDK and webrtc-streamer code.
However this should not affect running with docker (that contains a enouph recent GLIBCXX).

Best Regards,
Michel.

mpromonet, 2020-08-09:
Hi obalyuk87 ,

It should works back with latest docker.

Best Regards,
Michel.

closed 2020-08-31 by mpromonet