mpromonet/webrtc-streamer #345: Single CPU pegged at 100% by webrtc-streamer

D1plo1d, 2020-08-11:
**Describe the bug**
I'm using Webrtc-streamer to stream feeds from 3D printers and I've noticed that some times CPU usage for reasons unknown goes to 100% and stays there.

If it helps there may have been an undervolting of the camera around the time that the problem happened - so possibly the webcam disconnected and reconnected many times?

I'm not really sure what's going on here so my question is primarily what are my next steps to debugging this / filing a better bug report?

**Screenshots**

[image]

**Desktop (please complete the following information):**
- OS + Hardware: Raspbian / Raspberry Pi 4
- Browsers: Firefox + Chrome

mpromonet, 2020-08-12:
Hi D1plo1d,

In such a situation you can run `strace -p <thread pid>` to better understand what is doing the thread.

Best Regards,
Michel.

D1plo1d, 2020-08-13:
Thanks Michel, I'll try that.

greendrake, 2020-12-28:
Having this problem too — CPU usage is too high on Linux. A few seconds of `strace` shows:
```
% time     seconds  usecs/call     calls    errors syscall
------ ----------- ----------- --------- --------- ----------------
 86.00    0.916980         314      2914           epoll_wait
  9.35    0.099728           8     11864       133 futex
  4.64    0.049432          17      2898           read
  0.01    0.000099           6        16           write
------ ----------- ----------- --------- --------- ----------------
100.00    1.066239                 17692       133 total
```

The trace shows heaps of `FUTEX_WAKE_PRIVATE` lines. The process spawns itself into about 10 subprocesses (maybe this is fine though, I don't know).
Any ideas what might be wrong?