mpromonet/webrtc-streamer #346: Question on Embed in a HTML page or WebComponent with a V4L2 capture device or Raspberry Pi Camera

lukelu0520, 2020-08-12:
Dear webrtc-streamer friends,

Thanks for the webrtc tool, so far it works nice to us. We are integrating it to html with our raspberry pi camera (mmal service 2016.1). One thing confuses us is the url setting for HTML embedding and WebComponent.
In the Readme, it gives examples using rtsp source such as:
 
_webRtcServer.connect("rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov");_
or
_<webrtc-streamer url="rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov"></webrtc-streamer>_

When we set url to _http://webrtc-stream-hostname:port/webrtcstreamer.html?video=mmal%20service%2016.1_ for raspberry pi camera, it doesn't work. (directly use the address in browser would work)

So we were wondering what's the proper way to set url for HTML embedding or webcomponent.

Also, Does the .html page server and the webrtc-stream server need to be in the same machine or in the same local network ?

-- Luke

mpromonet, 2020-08-12:
Hi Luke,

The example in the README use rtsp backend url but it could be whatever webrtc stream name, it could be screen://, or a capture device.

The 2 ways to embeded was 2 differents experimentation, the webcomponent was to explore the feature. This approach is nice by its HTML simplicity.
You could find some try to use the webcomponent in vuejs https://github.com/mpromonet/webrtc-streamer-vuejs.

About the last point, the html/js could be served by an other machine on another network, the url to reach the api just need the way to reach it.

Best Regards,

Michel.

lukelu0520, 2020-08-12:
Hi @mpromonet thanks for your reply.

We were wondering if you have met cross origin issue when calling resource in port 8000 (webrtc stream default port) from http default port (e.g. 80).

Cross-Origin Request Blocked Error like below:
> Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://local:8000/api/addIceCandidate?peerid=0.23490973087995082. (Reason: CORS header ‘Access-Control-Allow-Origin’ missing).

Hope there is a way to relax the restrict by setting up something in webrtc-streamer config.

-- Luke

vegalou, 2020-08-13:
R-Proxy webrtc-streamer:8000 to your httpd apache/nginx to skip CORS error.

mpromonet, 2020-08-13:
Hi,

The CORS header `Access-Control-Allow-Origin: *` is set in all api of webrtc-streamer, but if you proxy it the header may be lost.

Best Regards,
Michel.

vegalou, 2020-08-13:
#### mpromonet

Reversed. **;-)**

I guess luke embedded webrtc-streamer's httpd(`:8000`) in REAL web daemon in 80/443.

Regards.

lukelu0520, 2020-08-13:
Thanks for @vegalou and @mpromonet , your active supports are really helpful.

The html is running now when we stream our own source s.t. screen://... or windows://..., though the error message (including Cross-Origin Request Blocked and the message in #334) would still show up in browser console. (we are using Chrome version 84.0.4147.105 (Official Build) (64-bit) and Firefox 79.0 (64-bit) )

Regarding streaming third-party source,e.g. from rtsp://wowzaec2demo.streamlock.net , we sometimes (~1 out of 10 trials) can get the streaming working, but the success is pretty random without the reproducibility.

We'll take your suggestion to see if we can find the root cause and remedy it.

Best,
-- Luke

closed 2020-09-01 by lukelu0520