mpromonet/webrtc-streamer #347: H265 RTSP to webrtc-streamer

vegalou, 2020-08-13:
Try to open h265 IP camera streaming to webrtc-streamer, 
The streaming works via VLC(valid), seems codec not included.

Or more a ffmpeg transcode to RTSP server in H264 preparation.

Thanks to @mpromonet !

mpromonet, 2020-08-13:
Hi @vegalou,

The decoding is based on the WebRTC classes, in particular, it use the factory `webrtc::InternalDecoderFactory` :
```
std::vector<SdpVideoFormat> InternalDecoderFactory::GetSupportedFormats()
    const {
  std::vector<SdpVideoFormat> formats;
  formats.push_back(SdpVideoFormat(cricket::kVp8CodecName));
  for (const SdpVideoFormat& format : SupportedVP9DecoderCodecs())
    formats.push_back(format);
  for (const SdpVideoFormat& h264_format : SupportedH264Codecs())
    formats.push_back(h264_format);
  if (kIsLibaomAv1DecoderSupported)
    formats.push_back(SdpVideoFormat(cricket::kAv1CodecName));
  return formats;
}
``` 
Then a way to implement H265 decoding is to extend this factory.

Best Regards,
Michel.

vegalou, 2020-08-14:
Best! @mpromonet 

Seems a though. `webrtc suppport h265` #G

closed 2020-08-14 by vegalou