mpromonet/webrtc-streamer #348: Problems on cross-server

lkf123456, 2020-08-18:
Current Setup:

- Server A: Only runs webrtc streamer
- Server B: HTTP which only has a simple html embedding the WebRtcStreamer as shows in the example web page for the time being.

Both servers are in the same network.

I tried open the web page from server B itself, so essentially the Chrome is accessing localhost (server B) and sending request to server A, and installed [Allow Cors access control plugin](https://chrome.google.com/webstore/detail/allow-cors-access-control/lhobafahddgcelffkeicbaginigeejlf) plugin due to cors issue, but it still didn't succeed. Here is the console log:

```
oniceconnectionstatechange  state: checking
webrtcstreamer.js:286 addIceCandidate OK
webrtcstreamer.js:160 oniceconnectionstatechange  state: disconnected
```

So I decided to move the HTML to server A, and access from server B, and of course installed the plugin, but the following happens:
```

adapter.min.js:5 Uncaught (in promise) TypeError: Failed to execute 'addIceCandidate' on 'RTCPeerConnection': Candidate missing values for both sdpMid and sdpMLineIndex
    at RTCPeerConnection.withCallback [as addIceCandidate] (adapter.min.js:5)
    at WebRtcStreamer.onReceiveCandidate (webrtcstreamer.js:289)
    at webrtcstreamer.js:138
    at tryCallOne (request.min.js:10)
    at request.min.js:10
    at MutationObserver.flush (request.min.js:27)
```

The only way it get works is to access the web page from server B itself.
Any solutions to this?

andrewawni, 2021-01-13:
@lkf123456 I'm also facing a similar issue, have you managed to solve this?

mpromonet, 2021-01-31:
Hi,

I am not sure what is your problem.

Using following html served by a http server (I tried serving it with python): 
```
<html>
<head>
    <script type="module" src="https://webrtc-streamer.herokuapp.com/webrtc-streamer-element.js"></script>
</head>
<body>
   <webrtc-streamer url="Bunny" webrtcurl="https://webrtc-streamer.herokuapp.com"></webrtc-streamer>
</body>
</html>
```
Show the Bunny test on the page.

Best Regards,
Michel.

BQSQ, 2021-02-24:
I'm also facing a similar issue,   
after testing,
chromium :88.0.4324.190 - happens

chromium70.0.3538.25 -  ok

mpromonet, 2021-03-13:
Hi @BQSQ, @lkf123456, @andrewawni 

I don't understand the original problem, that may differ of people that seems to think sharing a common problem.
Please open a well described problem if you like to describe an issue.

Best Regards,
Michel.

closed 2021-03-13 by mpromonet