mpromonet/webrtc-streamer #350: is ther a headless TURN based webcam with mic enabled example?

Joel-Mckay, 2020-09-01:
Hello,
Like many other tickets/sites I read, I too ran into Audio stream access issues in both FF and Chromium.
Not a lot out there about how the config.json file should be configured, so my stream setup attempt may indeed be borked.
=)
Cheers,
J

To Reproduce, one would require several steps to build, but I think it is probably something trivial I missed in reading the  javascript api access calls.  There are 3 different issues I encountered with a headless build of the webrtc lib and webrtr-streamer audio.

The first issue is related to starting from a ssh session, but I will hopefully be able to document that issue myself when I get any audio at all working first. I will include the error text here for other searchers in the future:
```
  [000:007] [16643] (audio_device_impl.cc:341): Audio device initialization failed.
  Fatal error in: ../../media/engine/adm_helpers.cc, line 39
  last system error: 88
  Check failed: 0 == adm->Init() (0 vs. -1)
  Failed to initialize the ADM../run.sh: line 6: 16643 Aborted                 (core dumped) ./webrtc-streamer -s- -T0.0.0.0:3478 -tturn:turn@$(curl -s ifconfig.me | grep -m1 -oE '[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}' ):3478 -vvv -a0 -C config_vid0.json -R 3460:3570
```

The second issue was when removing the muted flag from a video tag caused the frame to freeze in firefox 79 amd64 Specifically I edited html/webrtcstreamer.html in the example so it would enable audio after the main panel tile in index.html was clicked (i.e. hopefully it would help people stop filing tickets for this same issue if it is automatically activated after a user action).. 
 
The third issue is the audio access in the headless build.
 My **CMakeLists.txt** webrtc lib build flags for headless OS webcam:
`  set (WEBRTCARGS rtc_include_tests=false\nrtc_enable_protobuf=false\nrtc_include_pulse_audio=true\nuse_ozone=false\nrtc_use_x11=false\nrtc_use_x11_extensions=false\nrtc_use_gtk=false\nrtc_disable_trace_events=false\nrtc_disable_logging=false\nis_unsafe_developer_build=false\nis_desktop_linux=false\nrtc_build_examples=false\nrtc_build_tools=false\ntreat_warnings_as_errors=false\nrtc_enable_libevent=false\nrtc_build_libevent=false\n)
`
API reported devices:
```
  api/getVideoDeviceList
  [
	"UVC Camera (046d:09c1)"
  ]
```

```
api/getAudioDeviceList
[
	"default: QuickCam Deluxe for Notebooks Analog Mono",
	"Built-in Audio Analog Stereo",
	"QuickCam Deluxe for Notebooks Analog Mono"
]
```
```

api/getPeerConnectionList  shows the audio stream should be there (even if no sound is played)...
[
	{
		"0.21728792840431377" : 
		{
			"ice_state" : 2,
			"pc_state" : 2,
			"sdp" : ...
			"signaling_state" : 0,
			"streams" : 
			{
				"vid0|vid0|" : 
				{
					"vid0|vid0|_audio" : 
					{
						"kind" : "audio",
						"state" : 1
					},
					"vid0|vid0|_video" : 
					{
						"height" : 120,
						"kind" : "video",
						"state" : 1,
						"width" : 160
					}
				}
			}
		}
	}
]
```

Stream setup file:
```
cat config_vid0.json 
{"urls":{
        "vid0": {"video": "UVC Camera (046d:09c1)", "audio": "QuickCam Deluxe for Notebooks Analog Mono"}
 }}

```
Webcam TURN setup:
`./webrtc-streamer -s- -T0.0.0.0:3478 -tturn:turn@$(curl -s ifconfig.me | grep -m1 -oE '[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}\.[0-9]{1,3}' ):3478 -vvv -a0 -C config_vid0.json -R 3460:3570
`
 
I have attached the log file for the stderr output.

[bug.txt](https://github.com/mpromonet/webrtc-streamer/files/5153271/bug.txt)

Joel-Mckay, 2020-09-01:
I attached the new  **html/webrtcstreamer.html** I would recommend for the examples. It works in an X11 user login session for my cam's mic stream, but I am getting intermittent clicks which I assume is the talk detection filter in the webrtc lib itself.
[webrtcstreamer.html.zip](https://github.com/mpromonet/webrtc-streamer/files/5158064/webrtcstreamer.html.zip)

As my other OS specific issues are unrelated to the **webrtc-streamer** project itself... I am closing my ticket. 

Cheers,
J

Joel-Mckay, 2020-09-01:
stay awesome ;-)

closed 2020-09-01 by Joel-Mckay

mpromonet, 2020-09-02:
Hi Joel,

Then it works need a kind of interactive ssh, it may have some ALSA configration to avoid this.
For your modification in webrtcstreamer, you may submit a pull request to https://github.com/mpromonet/webrtc-streamer-html.

Best Regards,
Michel.