mpromonet/webrtc-streamer #351: How start docker image with active turn server?

shurshilov, 2020-09-15:
I start my webrtc server as docker run -p 8000:8000 -it mpromonet/webrtc-streamer.
I have static Ip, and it work from internet, but localy no. 
I think I need a turn server but how to activate it in docker?
```
<html>
<head>
<script src="request.min.js" ></script>
<script src="adapter.min.js" ></script>
<script src="webrtcstreamer.js" ></script>
<script>        
    var webRtcServer      = null;
    window.onload         = function() { 
        webRtcServer      = new WebRtcStreamer("video","https://eurodoo.com:8000");
	webRtcServer.connect("rtsp://user:pass@s733.eyekraft.ru:36354/1/h264major");
    }
    window.onbeforeunload = function() { webRtcServer.disconnect(); }
</script>
</head>
<body> 
    <video id="video" />
</body>
</html>
```

mpromonet, 2020-09-16:
Hi,

Did you try with `--net=host` ?

Best Regards,
Michel.

shurshilov, 2020-09-16:
> Hi,
> 
> Did you try with `--net=host` ?
> 
> Best Regards,
> Michel.

Then I get 502 bad gataway

mpromonet, 2020-09-19:
Hi

> Then I get 502 bad gataway

Seems more an nginx error message than a error from webrtc-streamer.
You may look to that stuff....

Best Regards,
Michel.

shurshilov, 2020-10-04:
My setting was 8000 port on firewall proxy to 8001 nginx and than nginx encrypted using SSL and proxied to port 8000 on which the docker image was running. This scheme did not always work. For example, it worked through the network of a mobile operator, it was not inside my network, and for example, when I entered from a computer at work, it did not work either.

Further why I did it. I need to make a widget inside the site that displays the rtsp stream from the camera. The site is naturally https. If you make an iframe directly to the 8000 port of the docker, then since the docker image has http, the browser blocks such an iframe. So I had to proxy. 
Used iframe as js code doesn't work.

shurshilov, 2020-10-04:
[image]
this i got if call directly docker http

shurshilov, 2020-10-04:
I do it man))
Just change
//WRONG dont do it                        webRtcServer.connect(myurlvideo);
                        var url = {'video':this.value}
                        var options = "rtptransport=tcp&timeout=60";
                        webRtcServer.connect(url.video,url.audio,options);

mpromonet, 2020-10-09:
Hi,

Good to see that giving option argument solve your problem. 
This means that the default transport RTP/UDP between rtsp source and webrtc-streamer doesnot send stream.

Best Regards,
Michel.

closed 2020-10-09 by mpromonet