mpromonet/webrtc-streamer #353: No debug information is webrtc files

alexfreed58, 2020-09-18:
I have built the webrtc-streamer by setting WEBRTCBUILD=Debug. Then "cmake . " successfully built the executable. If I run it under gdb and set a breakpoint, I can see the line numbers, etc. for the streamer's own files, but the functions in the rest of webrtc are "black boxes".
 For example:
``
(gdb) where
#0  0x0000555556003294 in webrtc::PeerConnection::ApplyRemoteDescription(std::unique_ptr<webrtc::SessionDescriptionInter
face, std::default_delete<webrtc::SessionDescriptionInterface> >) ()
#1  0x0000555556002a4d in webrtc::PeerConnection::DoSetRemoteDescription(std::unique_ptr<webrtc::SessionDescriptionInter
face, std::default_delete<webrtc::SessionDescriptionInterface> >, rtc::scoped_refptr<webrtc::SetRemoteDescriptionObserve
rInterface>) ()
#2  0x0000555556025dc5 in rtc::rtc_operations_chain_internal::OperationWithFunctor<webrtc::PeerConnection::SetRemoteDesc
ription(webrtc::SetSessionDescriptionObserver*, webrtc::SessionDescriptionInterface*)::$_14>::Run() ()
#3  0x00005555560021ae in webrtc::PeerConnection::SetRemoteDescription(webrtc::SetSessionDescriptionObserver*, webrtc::S
essionDescriptionInterface*) ()
#4  0x000055555580ef91 in webrtc::MethodCall<webrtc::PeerConnectionInterface, void, webrtc::SetSessionDescriptionObserve
r*, webrtc::SessionDescriptionInterface*>::Run() ()
#5  0x000055555587164a in rtc::Thread::QueuedTaskHandler::OnMessage(rtc::Message*) ()
#6  0x0000555555870b5b in rtc::Thread::Dispatch(rtc::Message*) ()
#7  0x00005555558711ac in rtc::Thread::Run() ()
#8  0x0000555555797731 in main (argc=4, argv=0x7fffffffdd28) at /home/alex/streamer/webrtc-streamer/src/main.cpp:283
``
Only the main() has the debug info. How can I build with full debug including the webrtc code?

mpromonet, 2020-09-18:
Hi @alexfreed58 ,

In order to get everything in debug you should try to run: 
```
rm -rf ../webrtc/src/out
cmake -DWEBRTCBUILD=Debug .
make
```

Best Regards,
Michel.

alexfreed58, 2020-09-18:
I did just that and found what is probably a build system bug that caused my problem in the first place. After deleting the out directory I ran  cmake -DWEBRTCBUILD=Debug . which re-created the out/Debug directory, however the "make" failed because it wanted to link with **Release**/obj/libwebrtc.a
A quick investigation found a reference to that in 2 places: the CMakeCache.txt and  build.make
After manually changing that to Debug, I was able to build.

I suspect that before I was *thinking* a Release version was build due to the -DWEBRTCBUILD=Debug flag, but it was actually linking with the Release library that also existed, so there was no error reported.

mpromonet, 2020-09-18:
Yes sure, you should delete *.cmake CMakeCache.txt CMakeFile.dir....

alexfreed58, 2020-09-18:
Thanks for your help. Now if I could just get rid of literally hundreds of warnings like:
warning: Could not find DWO CU obj/third_party/jsoncpp/jsoncpp/json_reader.dwo(0xd99a2baae3721763) referenced by CU
at offset 0x217624

closed 2020-09-22 by mpromonet