mpromonet/webrtc-streamer #354: Can't get a stream FROM the browser.

alexfreed58, 2020-09-22:
I'm trying to get a stream in the other direction as well.  I have added this to the webrtcstreamer.js file:
 navigator.mediaDevices.getUserMedia(mediaConstraints).then(function (stream) {
		    global_pc.addStream(stream)});
Now when I get to the streamer's page (via https) the camera lights up, but apparently no stream is actually sent. I can see in webrtc-internals, that there a 2 sender streams, video and audio, but nothing gets sent:

Statistics RTCMediaStreamTrack_sender_2
--
timestamp | 9/21/2020, 5:12:31 PM
trackIdentifier | 26c3bf8c-ecd7-4145-ac33-90a616c6d4ae
mediaSourceId | RTCVideoSource_2
remoteSource | false
ended | false
detached | false
kind | video
frameWidth | 0
frameHeight | 0
framesSent | 0
[framesSent/s] | 0
hugeFramesSent | 0

What am I missing? Was the answer not formed correctly? Wrong constraints?  Please advice!

mpromonet, 2020-09-22:
Hi,

You may look to https://github.com/mpromonet/webrtc-streamer-html/blob/master/sendstream.html that send a stream to the streamer. By now the implementation of receiver just print some stats.

Best Regards,
Michel.

alexfreed58, 2020-09-22:
Thank you, Michel, but could you please elaborate a little. Webrtc is very new to me. If I open the page "https://192.168.1.2:8443/sendstream.html?video=myvideo&audio=myaudio" in the browser, I get local video in a window  and Wireshark shows that indeed there is a stream from the browser to the "streamer".  Better yet I have changed the function 
PeerConnectionManager::VideoSink::OnFrame() and I can get to the actual video frame. Great! Now one more question. This way I only SEND the video from the browser, but not get anything back, while using the index.html I was getting video from the server. I suspect I only need a few lines of JS to create a bidirectional connection, after all "sendstream.html" defines remotevideo. What should I add (to webstreamer.js ?) to make it happen? Thanks again.

mpromonet, 2020-09-23:
Hi,

You should get (myvideo, myaudio) WebRTC stream. 

Best Regards,
Michel.

alexfreed58, 2020-09-23:
I am getting the (myvideo, myaudio) WebRTC stream. No problem. But usually the webrtc connections are two-way. What needs to be in the JS code to both send and receive video?

mpromonet, 2020-09-23:
Hi,

It does, I made some fix to display the remote stream.

Best Regards,
Michel.

closed 2020-09-24 by alexfreed58