mpromonet/webrtc-streamer #355: docker pull failed

hp6120c, 2020-09-27:
when I run `docker run -p 8000:8000 -it mpromonet/webrtc-streamer` in my server(ubuntu 18.04), It is failed and give me a message:`docker: error pulling image configuration: unknown blob` ,I dont know how to fix it
`

mpromonet, 2020-10-04:
Hi,

The latest dockerhub image is corrupted https://hub.docker.com/r/mpromonet/webrtc-streamer/tags 
`d59494321eca
linux/amd64
0 B`

You should try the lastest tag:
`mpromonet/webrtc-streamer:v0.3.2
`

I will regenerate the image.

Best regards,
Michel.

lucastcox, 2020-10-06:
Thanks, @mpromonet, just ran into the same issue.

mpromonet, 2020-10-07:
@lucastcox 

latest should be ok now since commit https://github.com/mpromonet/webrtc-streamer/commit/f6451624ad48e3000cf7141d6b74d07880a790ea

Best Regards,
Michel.

closed 2020-10-09 by mpromonet