mpromonet/webrtc-streamer #358: Many streams from one source

alexfreed58, 2020-10-08:
I have a question. If I understand correctly, if several clients want to get a feed from the same source, say a camera, the streamer uses a single CAPTURE input, but a separate encoder for each client. Since video encoding is a CPU intensive process, I'm wondering if it is possible to encode the stream once but use the same EncodedImage for multiple streams.

mpromonet, 2020-10-09:
Hi,

This is done by the WebRTC SDK, but my understanding is that each client have a different negotiation to choose the format, resolution and dynamically adapt the bandwidth, then the encoding is different for each client.

Best Regards,
Michel.

alexfreed58, 2020-10-09:
Thanks for the answer, but do you have any suggestion as to how to approach this with the SDK?

mpromonet, 2020-10-10:
Hi,

You can submit your wishes to https://bugs.chromium.org/p/webrtc/issues/entry

Best Regards,
Michel.

alexfreed58, 2020-10-12:
I'm closing the "issue" as it was really a question rather than an issue :)

closed 2020-10-12 by alexfreed58