mpromonet/webrtc-streamer #359: Stream Works on index, but not when embedded in Webpage?

djn10city, 2020-10-11:
Hello,

I'll default that I may not be doing something correctly, but when I load the stream from the terminal or from index using /?<stream>, it loads fine.

When I use the example embed html and replace your stream with my stream, it doesn't load.

```
<html>
<head>
<script src="libs/request.min.js" ></script>
<script src="libs/adapter.min.js" ></script>
<script src="webrtcstreamer.js" ></script>
<script>        
    var webRtcServer      = null;
    window.onload         = function() { 
        webRtcServer      = new WebRtcStreamer("video",location.protocol+"//"+window.location.hostname+":8000");
	webRtcServer.connect("rtsp://74.94.110.81:7447/n9HSYhAtyQXiRTon");
    }
    window.onbeforeunload = function() { webRtcServer.disconnect(); }
</script>
</head>
<body> 
    <video id="video" />
</body>
</html>
```

Thoughts?

Thanks!

mpromonet, 2020-10-11:
Hi,

The index.html use the options defined in https://github.com/mpromonet/webrtc-streamer-html/blob/master/webrtcstreamer.json, so maybe you need to add option "rtptransport=tcp" and connect using : 
`webRtcServer.connect("rtsp://74.94.110.81:7447/n9HSYhAtyQXiRTon", "", "rtptransport=tcp");`

Best Regards,
Michel.

djn10city, 2020-10-12:
Beautiful.  Worked like a champ.

Thank you!

closed 2020-10-13 by mpromonet