mpromonet/webrtc-streamer #360: CMake Error: The following variables are used in this project, but they are set to NOTFOUND.

httpsphp, 2020-10-23:
Hello,

I am new to WebRTC and I'm trying to implement your RTSP - WebRTC Steamer.

I am having some troubles with the CMake, I receive this error msg:

`WEBRTCBUILD = Release
WEBRTCROOT = /opt/webrtc-streamer/../webrtc
WEBRTCDESKTOPCAPTURE= ON
CMAKE_CXX_COMPILER_ID=GNU
ALSA_FOUND = FALSE
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
WEBRTC_LIBRARY
    linked by target "webrtc-streamer" in directory /opt/webrtc-streamer

-- Configuring incomplete, errors occurred!
See also "/opt/webrtc-streamer/CMakeFiles/CMakeOutput.log".
See also "/opt/webrtc-streamer/CMakeFiles/CMakeError.log".
root@webrtc-sol:/opt/webrtc-streamer#
`

I have tried editing CMakeLists.txt to(commented find_package) but I get the same error:
`set(CMAKE_THREAD_PREFER_PTHREAD TRUE)
set(THREADS_PREFER_PTHREAD_FLAG TRUE)
# find_package (Threads)
target_link_libraries (${PROJECT_NAME} Threads::Threads) 
`

Upon reading CMakeError.log, I found this issue here:
`CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'`

GNU Make 4.1
Ubuntu 18.04.5 LTS
libpthread-stubs0-dev

What can I do to get this working? Is there something Im missing? Thank you

mpromonet, 2020-10-24:
Hi,

Maybe like to the arch you are using ? 

The first error say WebRTC is not present, did you download it using `fetch` depo_tools command in folder `/opt/webrtc-streamer/../webrtc` ?

Best Regards,
Michel.

ticks0628, 2020-11-19:
I encounter your situation also. CMake can't reference pthread. It's strange and totally irrelevant.
You just need to download live555 and civetweb manually then you can build it successfully.

Ubuntu 20

httpsphp, 2020-11-19:
@mpromonet and @ticks0628  - Thanks for both replies. I was able to get this successfully working like a charm, but after some research, I found that Kurento Media Server better suits my project. This can be closed.

mpromonet, 2020-11-19:
Hi httpsphp,

Kurento is quite different and have more latency.
There is others solutions to make gateway between RTSP & WebRTC like janus-gateway, flashphoner,....

Best Regards;
Michel.

closed 2020-11-29 by mpromonet

hossem1ch, 2020-12-10:
@httpsphp  try to upgrade your gcc to 7.5.0 version